simplefoc/Arduino-FOC #2: Motor Reversing/Direction change

MatthewReed303, 2020-04-11:
Hi, Is there any way to reverse/change the direction of the motor? I'm also looking at using it on an stm32f103 blue pill, could the library be used with stm32? From what I can see is only the PWM high frequency function would need to be changed in order to use the stm32 clocks. I would like to use external mosfets for larger loads so that's something I will also look into. Will you implement using halls instead of the encoder? I plan on running my encoders directly to my CNC controller rather than being used for motor commutation, but I could possibly run encoder to both. Basically I want to build little FOC BLDC speed controllers using low cost blue pills for driving BLDC motors for servo control applications hobby CNC stuff using BLDC motors rather than steppers.

askuric, 2020-04-11:
Hello Matthew, 
Thanks for the message.

> Is there any way to reverse/change the direction of the motor?
- If you are thinking of controlling the motor in different directions, it certainly can change the direction. :)
At this point you use few different control loops angle, velocity and ultra slow velocity loop. In all of these loops you can change the direction of the motor. 

- If you were thinking of the reversing the motor in terms of generating energy, that will depend on your driver. But the library gives you the possibility to set negative voltages (if you hardware allows it) on the motor. 

> I'm also looking at using it on an stm32f103 blue pill, could the library be used with stm32?

I have not tried this library on the bluepill but I am eager to do it and I hoe to test it next week.

> From what I can see is only the PWM high frequency function would need to be changed in order to use the stm32 clocks.

If you would like to use this code as is. Using Arduino IDE than it will need some minor adjustments, like this one. for sure. 
But if you would like to do it using CubeMX and Keil IDE than you will need some more adjustment. Mostly in terms of pwm timers and encoder counters. 
Notrhing to major, I would be interested to help you with it as well. 

> I would like to use external mosfets for larger loads so that's something I will also look into. Will you implement using halls instead of the encoder?

Definitely a way to go, and a future (near) extension.

> I plan on running my encoders directly to my CNC controller rather than being used for motor commutation, but I could possibly run encoder to both. 

I am not sure that I understand completely, but to run the FOC you will need a certain way to retrieve the angle measurement. It can be using any sensor/communication you want. At the moment this simple library uses only encoders directly but it can be easily extended to use different sensors or even communication interfaces. 

> Basically I want to build little FOC BLDC speed controllers using low cost blue pills for driving BLDC motors for servo control applications hobby CNC stuff using BLDC motors rather than steppers. 

That is a really cool application. Feel free to contact me directly if you want to discuss this in depth. 
Good luck!

MatthewReed303, 2020-04-11:
Hi Antun,

Can you please provide a simple example of changing the direction, is this just a function that can be called in the loop or is there a bit more to it? 

I will try it with an Uno first and get it all working and then I will move to blue pill and see how I go. I don't think it will require too much if I stick to using the Arduino Boot loader on it as I currently do.  But I will keep you updated and If I succeed I will pass on the working code for you to add if I get it done before you have a chance. 

I will do some testing tomorrow using ir2101 to drive the MOSFETs.
http://grauonline.de/wordpress/?page_id=3122 I will be using these cheap Chinese boards and removing there JY01 IC. It's very noisy control and has a soft start that can't be switched off, the reversing ramps motor down and then ramps it back up which is hopeless for CNC applications. The CNC motion controller will control acceleration and deceleration and needs no manipulation from the BLDC controllers or it's impossible to tune the PID loop in. I currently have the CNC controller outputting 10khz PWM and a direction bit. This works great with the A4915 https://www.allegromicro.com/en/products/motor-drivers/bldc-drivers/A4915 the only problem I have is I can't get low speeds or it becomes very unstable as that driver uses trapezoidal control. From my research, I need sinusoidal or FOC for smooth operation at low speeds. 

https://www.infineon.com/dgdl/Infineon-ir2101-DS-v01_00-EN.pdf?fileId=5546d462533600a4015355c7a755166c 
The only problem I see with using the IR2101 is they use HIGH IN and LOW OUT to drive the 2x Mosfets for the H bridge. I have gone through your code and don't see this type of configuration. I take it this logic must be done in the L6234 Driver you are using. How would I go about using your code to drive these, I know the PWM out will be on the high side but will need to add some logic for the low side. What would be your recommendations or is there already a way to do this? 

Thanks,
Matthew

askuric, 2020-04-11:
Hey Matthew,

This is starting to be a multi-front communication. So let me answer you in short here and then I will send you some more details in an email. :)

I will add a simple example to the examples folder no problem. 
The reversing direction is done by providing the target velocity to the `motor.move()` funciton. It is very simple to do in fact. 

Regarding the ir2101 I have not taken in cosideration that case. But as I said this is a basic implementation at the moment and I am more than happy to extend it to your case as well.

Cheeres,
Antun

closed 2020-04-11 by askuric

valeriod, 2022-08-10:
>I will add a simple example to the examples folder no problem.
What's the name of the file showing the example of changing the direction?
Thanks