simplefoc/Arduino-FOC #4: Can I use this code to control brushed DC motors

Meekdai, 2020-04-27:
I have a project that needs to control 5 brushed DC motors with encoders. Of course, I need to control their absolute positions. Are there examples for reference? thanks~

askuric, 2020-04-27:
Hi,
The main point of this library is to use the Field Oriented Control (FOC) approach which doesn't make too much sense in terms of brushed DC motors. 
But you are right that it could be used in that sense as well, there is a lot of infrastructure that can be reused.  
At the moment it is not supported.

But it would not be too complicated to implment the DC motor support. 
If you are interested I can create you a branch with DC motor example implementation of the library.

Antun

Meekdai, 2020-04-27:
Thanks @askuric , I did not find library that support for DC Motor Closed Loop Control in the arduino community. So I found your library through FOC. If it can support DC motors, it's really great.

askuric, 2020-04-27:
Hey @Meekdai,
Thanks for the heads up. I will definitely think about it. I will look into it and maybe make an additional library for the DC's. 
Give me a day or two and I will send you a adaptation for the DC motors. 
At the moment I will not be able to test it properly, though so be aware.
Antun

Meekdai, 2020-04-27:
What a good news!  I will be very glad to test your new DC motors library.

askuric, 2020-04-27:
Hey @Meekdai , I have created a new repo where I will update the DC motor control library. 
https://github.com/askuric/Arduino-Closed-Loop-Motor-Control
At the moment there is only one example. 
Please check it out and test it when you have time. :D
I have checked only the compilation and it compiled well. 
Let me know what do you think.
Antun

Meekdai, 2020-04-28:
It's really fast. I may need some time to set up a test environment. I will do it as soon as possible. I closed this issue , and then will open new issue on the new library. thanks.

closed 2020-04-28 by Meekdai