simplefoc/Arduino-FOC #5: Slow velocity

raimapo, 2020-05-12:
Hey @askuric ,

I just finished to port Your library. It was some time while I have back and implemented microsecond timer. It solved problem. I have a few issues and comments at the same time.
1. At slow velocity motor behaves strangely. For example at 0.5 rad/s. Signal is quircky and rotation is by jumping. I think it's PI tuning needs ? I am not familiar yet with PID and and other stabilization methods.
2. I saw that different motors need different 3 phase connections. Is it possible somehow to solve that ? it gived some time to realize it and change wiring. While simplest sine wave do not need such changing, it's practically no matter how you connect phases to motor drive, just needs to give 120 degree difference in signal.
Maybe I am mistaken somewhere ?

askuric, 2020-05-12:
Hey @raimapo, 
Nice to hear from you!

1. That is right, you will have to tune your PID controller. 
In general for slower velocities you will have to raise the values of your P and I value. 
The values depend on your motor but the `P` value should be in between `[0.1 and 10]` and `I` value should be in between `[10 and 300]`. 
Also I am not sure which version of the library you have ported, but for slower rotations you should use the low pass filter.  In the library you can modify it by changing the low pass filter time constant `LPF_velocity.Tf`. The higher the TF value the smoother the velocity measurement will be, but too high value will make the controller unstable. 
So the  `Tf` value is in seconds so it should be in range `[ 0 to 0.1]`.
2. The motor and the sensor have to be aligned in order for the FOC to work. I could definitely automate this process, but in that case you would loose the notion of absolute left and right.( positive and negative) 
It is the question of the choice. 

Just to clarify:
All the motors have the same type of connections and are connected the same. But the motor and encoder/as5048 have to have the same positive and negative direction. If they don't, then the motor wont turn. 
You can either inverse A and B of your encoder or inverse the motor connector and everything will work. 

Let me know how it goes.

askuric, 2020-09-06:
Hi @raimapo, 
In the new library [version 1.5.0](https://github.com/simplefoc/Arduino-FOC/releases/tag/v.1.5.0) the PI controller is updated to the PID, so you'll have a bit more control over your motor. 

And the automatic sensor direction detection has been implemented so therefore I am closing this issue.

closed 2020-09-06 by askuric