simplefoc/Arduino-FOC #6: No current sensing?

cvabc, 2020-08-12:
I'm wondering why it's possible to do FOC without sampling phase current.

askuric, 2020-08-13:
Hey @cvabc 

this library has started as a reaction to very expensive and somewhat difficult to understand hardware+software combinations for running the BLDC motors such as ODrvive, Trinamic and similar.

If you take for example ODrive, it has two mode of operation, either BLDC motor (high power) or gimbal motor.
When you are using the gimbal motor mode, it does not use the current control loop. When I saw that, I told myself, if I can make the rest of the functionalities relatively implemented well, we could have the same level of control for the gimbal motors as the ODrive for much lower price and with customisable code :smiley: 
This is the first reason why I did not initially implement the current control loop.
The second reason is because the boards providing the measurement of the phase current are still somehow rare and I did not have any of them on my hands to try out and implement. Recently I have came across the aliexpress board with the DRV8302/8301 chip which is perfect for this purpose. [Aliexpress link](https://fr.aliexpress.com/item/4000126430773.html?spm=a2g0o.productlist.0.0.702a312aXmzuUK&algo_pvid=50131a88-ac88-4755-bb71-978c07ec461e&algo_expid=50131a88-ac88-4755-bb71-978c07ec461e-5&btsid=0b0a119a15957548552557385e6f5e&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

The last reason is maybe the most important one regarding your question.
Field Oriented Control algorithms usually have current control loop for two reasons:
- tracking of rotor magnetic field orientation
- torque control

##### Tracking of rotor magnetic field orientation
FOC was first developed as the powerful control method for the AC motors and there it is not enough to measure the rotor position to know the magnetic field orientation of the rotor generated by the induced currents. You need to measure the currents of the stator and check the influence of the rotor field on them. Basically without the currents you will not know good enough how to set the right angle of your stator field. 
The good thing about BLDC motors is that they have permanent magnets in the rotor and their rotor field is not influenced by the stator currents (for smaller motors). Therefore it is enough to measure the position of the rotor to know the field orientation, no need to measure the current.

##### Torque control 
The electric motor torque `T` is a value proportional to the current `I`
```cpp
T = K*I
```
therefore in order to control the torque applied by the motor you need to measure the current. But since the current measurement is very often unavailable for smaller motors I decided to go for a simpler solution.

If you take the example of a DC motor, it has the same proportional relation in between current `I` and the torque `T` but in most cases you don't measure the current of the DC motor but build the position and velocity control of the DC motor without using the current loop, without using the torque control. So since DC motor doesn't really need the torque control (current measurement) then the BLDC motor doesn't as well. 
- Unless you really need precise torque control
- Or you need to limit torque and current 

Therefore in this library in this stage, you can do a generalised torque control using voltage (here is a quick explanation how it works: [voltage torque control](https://docs.simplefoc.com/voltage_torque_control)), velocity and position control.  

The current control loop is very high in my priorities and it will be implemented in next release of the library.

Let me know what do you think!
Did you have something more specific in mind with your questions?

cvabc, 2020-08-13:
Thanks! That's a really nice explanation!
Actually, I have googled a lot about FOC without current sensing and people say that it's not possible.
Previously, I though that because gimbal motors are slow and back EMF can be ignored so that `I = k * U` (but we also have inductor ...).
If possible I don't want current sensing either, since we can use cheaper hardware without ADC, advanced timer, etc.
I prefer gimbal motor because it's possible to remove power MOSFET & driver. I would be happy if we can also remove ADC.

askuric, 2020-08-15:
Controlling the field orientation is certainly possible without measuring current. Because of the permanent magnets. There are some misaliglements that can happen due to difference dynamic effects, but in general you won't see the difference. There are also problems of FOC algorithms in for BLDC motors, one of them is for example assuming the currents for be sinusoidal. This is in many cases not true. They are more in a shape of trapezoidal waves. So sometimes an assumption like that can cause more problems than the misaliglement caused by not measuring the current. 

The problem that comes out not measuring the current is the restriction of motion control, not the restriction of FOC. You will not be able to control the torque directly and you will not be able to limit the current. But as we discussed before, this is true for most of DC motor applications not even to mention the stepper motors.

So using this library you will have a simple way to use sine of space vector modulation to motion control your BLDC motor, without the need for ADCs. Unless you want to control precisely the torque and if you want to limit the current.
Torque and current control applications were not the primary focus of the *SimpleFOClibrary* but are definitely going to be in future releases. 
And I am looking forward to see the results 😃

Koxx3, 2020-12-22:
still nope for current sensing ;)

askuric, 2020-12-23:
This proved itself to be a very difficult challenge. :D
Not really the foc part but the sensing part. We are working on this at the moment and we should have inline and low-side current sensing integrated soon.

Koxx3, 2020-12-23:
i know 😅😅
here is how stm32 does it on their firmware : 
https://community.st.com/s/question/0D50X0000B28qTDSQY/custom-foc-control-current-measurement-dma-timer-interrupt-needs-review

the pdf is very good to explain when/how to sample 😉

askuric, 2020-12-30:
Great link thanks!
I have seen already that stm32 boards are going to be the hardest to implement and especially when we want to have a good generic solution. I have so far implemented the DMA +ADC  and  ADC+interrupt so we will see which one will we use. 
I think for inline sensing DMA continious conversion will be a good option. But for low side sensing where the pwm needs to be in sync we will need an interrupt approach (with or without DMA). 

esp32 suprisingly has a pretty good support for the low-side current sensing off-the-shelf, the performance will maybe be the problem but the initial implementation works. It was already tested a bit by @byDagor.

Now in terms of Arduino devices, I think low side current sensing is off the table for the beginning. We will support only inline sensing for now. Or a very naive approach for low-side sensing (median filter).

Anyway, I hope to be integrating this soon!

runger1101001, 2023-03-21:
Current sensing is now supported on various MCU types:

STM32F1, F4, G4: low side and inline
ESP32: low side and inline
RP2040: inline
ATMega: inline
Teensy: inline
Due: inline
SAMD: low side and inline

runger1101001, 2023-11-28:
If it is ok, I will now close this issue since current sensing is now supported on quite a few MCU types...

closed 2023-11-28 by runger1101001