simplefoc/Arduino-FOC #7: SpaceVector is clipping (not centered)

owennewo, 2020-08-15:
The SpaceVectorPWM algorithm is clipping due to it not taking into account the voltage_power_supply.  

Expected behaviour:
 - the SpaceVector duty cycle for a,b,c will be centered on 0.5 duty cycle.  
Actual behaviour:
 - the duty cycle is down near zero and the signal is sometimes negative (which is then clipped)

This can result in stuttery movement or pulsing velocity.  Fix is pretty simple - I'll send a PR against this issue.
Here is an unlisted youtube video - showing my setup
https://www.youtube.com/watch?v=JMZoKTe1XRA

owennewo, 2020-08-15:
I've created PR #8 to fix

askuric, 2020-08-16:
Hey @owennewo,
Thanks for the awesome depth you are approaching this!

I have to go into this in depth because this part of the code is vital part of the FOC. This problem has arisen from something I did after the initial implementation. I am not sure what is causing it, but for now I am definitely happy to accept your pull-request. 

But I will leave this issue until I find the real reason why this is happening.

owennewo, 2020-08-16:
You take your time on finding a proper fix.
I didn't fully understand how SpaceVector was working but it is clear that as setPWM is dividing by voltage_power_supply then setPhaseVoltage needs to pass setPWM arguments in the range of 0 to voltage_power_supply.  
I think it might be easier to understand if setPWM received values between 0 and 1 and knows nothing of voltage_power_supply and all the voltage limiting scaling takes place in setPhaseVoltage.

askuric, 2020-08-16:
Yeah, actually I need to rethink this a bit. There have been many updates of this code so some of the key reasons of code structure have been a bit compromises.

For me `setPWM` is the one to handle all the limiting and if this fix turns out to be the ultimate solution, the centering and transformation to duty cycle. 
`setPhaseVoltage` should do pure modulation equations. 

At least in my mind, that was the idea.

But once I've looked from  `analogWrite` to duty cycle hardware specific code, this code became a bit harder to understand. 
Anyway, I am going to dig into this a bit deeper once I have the time and restructure the code, a bit better, for the next release.

askuric, 2020-09-06:
The solution to this issue has been integrated into the new new library release [v1.5.0](https://github.com/simplefoc/Arduino-FOC/releases/tag/v.1.5.0)

closed 2020-09-06 by askuric