simplefoc/Arduino-FOC #9: Control of hoverboard motors

happytm, 2020-08-16:
I would like to control 2 hoverboard motors for a custom wheelchair project using web interface . I already have simple web interface for ESP32 microcontroller  which is simple joystick which output 2 variables for reverse/forward & steering values based on joystick position.  I know there are lot of hoverboard firmware available on github but none of them are suitable for my hoverboard hardware. It uses Half bridge drivers based on IR2103/FD2103 chips . Following links describe my hardware:

Hoverboard:

https://www.ebay.com/itm/Bluetooth-Hoverboard-Self-Balancing-Electric-Scooter-LED-UL2272-CE-no-Bag-Sports/353089292564?ssPageName=STRK%3AMEBIDX%3AIT&var=622283901145&_trksid=p2057872.m2749.l2649%5D

Schematic:

[2.Mainboards_hoverboard_schematic.pdf](https://github.com/simplefoc/Arduino-FOC/files/5079779/2.Mainboards_hoverboard_schematic.pdf)

https://aidilj.com/2015/01/sensored-bldc-motor-control-with-arduino/

Is it possible to use your code with some modification to control these motors? 

Thanks.

owennewo, 2020-08-16:
From the schematic you provided your hoverboard is based on a gigadevice chip:
https://www.gigadevice.com/microcontroller/gd32vf103cbt6/
The hoverboards i have are based on the stmf103. Most of the hoverboard hacks  (and firmware) on the internet target stm32 .

That said the gd32 seems to be a clone of the st32:
https://smdprutser.nl/blog/stm32f103-vs-gd32f103-round-1-solderability/
So if you can work out how to upload firmware onto you might have a chance with existing firmware.
If you managed to do that, I'd say you'd also probably be able to port this library to the gd32.  

The library doesn't yet support hall sensors but this is on the roadmap.  I've added hall sensors support on one of my projects. 
I'm not sure how you see the esp32 fitting in. My suggestion is to either forget about esp32 and try to use gd32 working (using this library or other firmware) or ditch your board and build/buy an alternative.
Something very close to what you need might be the janus controller.  Talk to dgr on
https://community.simplefoc.com/
He's designed an esp32 board that uses this lib.
I've used the simplefoc board with a hoverboard motor.  
https://simplefoc.com/shop
This is great for experimenting with but i would only recommend it if you have a current limiting power supply otherwise you'll likely blow the board!.  I'd limit the current to <1A as the board has no mosfets.  You are looking for a board with 10A+ continuous.
One final thing about hall sensors and esp32.  The esp32 is 3.3v tolerant but the hall sensors on my esp32 required 5v (otherwise they would sometimes trigger multiple times).  I needed to do a voltage divider and a pullup resistor on each of the 3 hall sensors. Selecting the right resistors was harder than it sounds as they effect each other.

happytm, 2020-08-16:
@owennewo  Thank you for your detailed response. It seems like new efforts are made by yourself and dgr to make it work with these hoverboard motors with hall sensors. I will keep eye on couple of threads you and dgr started on forum.

> The library doesn't yet support hall sensors but this is on the roadmap. I've added hall sensors support on one of my projects.
I'm not sure how you see the esp32 fitting in. My suggestion is to either forget about esp32 and try to use gd32 working (using this library or other firmware) or ditch your board and build/buy an alternative.

My goal was to use existing hardware for motor driver and charging circuit to reduce cost as these board already have gd32 microcontoller, 3 half bridge drivers based on FD2103 chips and 6 high power mosfets per board.It also have MPU6050 chip for balancing but I do not need to use it as my project will have 2 extra pivot wheels. I am planning to use ESP32-CAM board to just send control commands to  GD32 boards and stream live video. So basically ESP32-CAM would work as wifi bride. following is example link:

https://robotzero.one/pan-and-tilt-control-for-an-esp32-cam/

Thank you.

owennewo, 2020-08-17:
So simplefoc supports some stm32 boards:
https://docs.simplefoc.com/microcontrollers

If you can work out how to flash your gd32 i suspect you could start to build your own firmware around simplefoc. It would be quite challenging but a great way to learn.  
Problems you may face include:
1. Flashing - i use platformio (instead of arduino ide) to develop and upload code. It supports a similar microchip:
https://docs.platformio.org/en/latest/boards/gd32v/sipeed-longan-nano.html
But your chip might not be easy to flash.
2. Driving the 3x FD2103 half bridges. 
The fd2103 isn't supported by SimpleFOC, it is possible that they are controlled in a similar manner to currently supported bldc drivers:
https://docs.simplefoc.com/bldc_drivers
This isn't my area of expertise.
3. Electronics schematics - you'll need to know where all the components are connected to gd32 and code for this
4. You won't get features for free.  E.g you'll have to implement things like detecting and beeping on low voltage. 
5. You'll need to come up with an interface between esp32 and gd32. E.g software serial or i2c. This will allow esp32 to send commands.

If this all sounds too complicated a slightly less difficult approach might be to buy a new hoverboard mainboard that is known to work with a hoverboard firmware.  Last time i looked they were about $20 on ebay.

Finally are you sure there are no firmwares for gd32?  E.g this claims to support it
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

happytm, 2020-08-17:
Thank you for your feedback. I think it is complicated to use existing drivers. The firmware you linked is for old style driver boards where there is one main board and 2 sideboards.

owennewo, 2021-01-02:
@happytm - is there anything we can do to help you on this issue.  Otherwise I'll close.

BTW: I've been using hoverboard motors with the STM g-b431g-esc1 board and it works reasonably well.  Not quite powerful enough but OK for my use case

happytm, 2021-01-02:
Thankyou for your support.

happytm, 2022-01-25:
It has been 1 year since last comment. I wanted to see if there is solution to upload SimpleFOC now on to above mentioned board.

runger1101001, 2023-10-31:
There is a thread on our forum where users are uploading to hoverboard drivers, and user @Candas has also developed a driver for some GD chips.
Please check our forum for details.

happytm, 2023-10-31:
@runger1101001 Thank you for your pointer.

Candas1, 2023-10-31:
Hi,

Several firmware were created in the past for hoverboard split boards (one controller for each motor).
[RoboDurden](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x) has forked one of the firmwares and is collecting pictures/pinouts about the different board layouts (there are many as you can see) to adapt the firmware via define.h files. I am sure he would be interested in those diagrams. I would suggest connect with him on that repo and check if you layout is covered. Unfortunately the gd32F130C6T6 chips don't have much memory so this could be a problem.

I worked on implementing drivers for gd32F130 so simplefoc could be used on those boards.
Robodurden's intention is to create a new/better [firmware](https://github.com/RoboDurden/Split_Hoverboard_SimpleFOC) for splitboards using SimpleFOC. It is still work in progress.

From my side I am concentrating on SimpleFOC as it will be beneficial to all the controllers.

happytm, 2023-10-31:
Hey @Candas1 ,

I have couple of hoverboards with GD32E230C8T6 (LAYOUT TYPE 2.7). I checked your repo but this layout isnot supported yet.It will be appreciated if you add support for Layout type 2.7 boards.

Thanks.

Candas1, 2023-10-31:
Read my answer, and talk to Robodurden....

happytm, 2023-10-31:
Thanks @Candas1 .

Candas1, 2023-10-31:
So you talked to him [here](https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x/issues/16#issuecomment-1774009616), he shared a binary, and you haven't provided feedback ? What do you expect ?

happytm, 2023-10-31:
I could not upload .bin file to hoverboard and then I saw this option here for possibly using SimpleFoc. I am more comfortable with Arduino.

Thanks.