simplefoc/Arduino-FOC #14: esp32_position_control example does not compile for ESP32 Wrover Module

josefinHeilig, 2020-08-31:
Hello,
I was using the Arduino-FOC with the STM32 which worked nicely.
Now I set it up with an ESP32 Wrover (board version 1.0.4), trying both the current master-branch and dev-branch (31.08.20) and it does not compile. the  ...\examples\hardware_specific_examples\ESP32\magnetic_sensor\esp32_position_control .

Arduino throws the following error message:

`C:\Users\Josefin\Documents\Arduino\libraries\Arduino-FOC-dev\src\FOCutils.cpp: In function 'void _setPwmFrequency(int, int, int)':

C:\Users\Josefin\Documents\Arduino\libraries\Arduino-FOC-dev\src\FOCutils.cpp:108:55: error: 'MCPWM_SELECT_SYNC_INT0' was not declared in this scope

   mcpwm_sync_enable(m_slot.mcpwm_unit, MCPWM_TIMER_0, MCPWM_SELECT_SYNC_INT0, 0);`

How can I fix this?
Thanks for your help!

josefinHeilig, 2020-08-31:
I received the simple answer on the Community Forum https://community.simplefoc.com/t/esp32-position-control-example-does-not-compile-for-esp32-wrover-module/92 from DGR:

> Easy fix! You just need to replace a file in your arduino installation and it will compile.
> 
> Here https://docs.simplefoc.com/microcontrollers you can find the steps in the ESP32 section.

closed 2020-08-31 by josefinHeilig

zhuangfengzi, 2020-10-19:
@josefinHeilig 
Thanks!