simplefoc/Arduino-FOC #21: Inverse Park and Clark are being shifted incorrectly

owennewo, 2020-09-15:
I've spent 2 or 3 hours on this and I think I've got to the bottom of it (at least for `SinePWM`).  The issue results in clipping of sinusoid PWM when `Uq` approaches `voltage_power_supply`.  It's easier to explain when considering `ControlType::voltage` 

I'll set out a scenario:
 * motor is in `ControlType::voltage`
 * with `voltage_power_supply = 9`
 * and `voltage_limit = 9`
 * and we asking for a voltage of `motor.move(9)`

In the above case I'd expect to see a beautiful sin wave (PWM'd) being fed to the three phases between 0 and 1.0 duty cycle.  What I'm seeing is heavy clipping.  Essentially dc_a,dc_b,dc_c reach a peak of ~13.5v and are being constrained (clipped) not to go over voltage_power_supply.  

To remove this clipping I would change:
```
// Inverse park transform
Ualpha =  -_sin(angle_el) * Uq/2; 
Ubeta =  _cos(angle_el) * Uq/2;  

// Clarke transform
Ua = Ualpha + Uq/2;
Ub = -0.5 * Ualpha  + _SQRT3_2 * Ubeta + Uq/2;
Uc = -0.5 * Ualpha - _SQRT3_2 * Ubeta + Uq/2;
```
Change1: `Uq/2` for both `Ualpha` and `Ubeta` (i.e. `Ualpha` and `Ubeta` are halved)
Change2: add `Uq/2` instead of `voltage_power_supply/2` to  for `Ua`, `Ub`, `Uc` terms (i.e. changing the amount we 'shift up') in order to get the sine wave sitting exactly on zero.

Apart from empirical evidence (no clipping observed), my thinking is this.
In voltage control `Uq = 9` (in above scenario)
`Ualpha` and `Ubeta` are both therefore in the range of `-9v to 9v`

So we are dealing with peak to peak of 18v not 9V!

If we change `inverse park transform` as code block above then range becomes `-4.5 to 4.5V`.  This works well with the zero shift below.

In `inverse clark transform` the old code would add `voltage_power_supply/2` to all `Ua Ub` and `Uc` terms meaning their range would shift to `0 to 9V`.  

Sounds good, however we really need to be shifting by `Uq/2` not `voltage_power_supply/2`. (its hard to see here because `Uq==voltage_power_supply` in above scenario)   If we consider another scenario where we are calling `move(3)` (i.e. Uq=3) with same `voltage_power_supply = 9`.  `Ua`, `Ub` and `Uc` before shifting would be `-1.5 to +1.5` sine waves.  Shifting by `voltage_power_supply/2` would raise it to 3 to 6v.  Shifting by `Uq/2` would raise to 0 to 3V.  We want the latter!

I realise this a complex area, and although I think I'm right, I may have made a mistake!

I also think `FOCModulationType::SpaceVectorPWM` is similarly incorrect but lets focus on this case first!

owennewo, 2020-09-16:
Right,  I'll have another go at this.
I think @askuric has already tried to explain to me that talking about voltage is ambiguous.  Do we mean DC or AC Peak or AC Peak to Peak or AC RMS.  

Looking at the main voltage variables and the library implementation, I'd say that:
- `voltage_power_supply` - is a DC voltage (that was easy)
- `Uq` - I'm not sure if this is best described as a DC voltage or as the amplitude of the AC voltage (i.e. AC Peak voltage), perhaps it doesn't matter
- `Ualpha, UBeta` and - `Ua, Ub, Uc` - are all AC and vary from `-Uq to +Uq`.  
We then see a shift in `Ua, Ub, Uc` so the range is more like a sine from `0 to 2Uq`.  
Finally we divide by voltage_power_supply. and constrain to 0.0 and 1.0 ready for PWM

What I'm seeing is if `Uq > voltage_power_supply/2` then the the sin wave gets clipped.
I think we should be limiting `Uq to < voltage_power_supply/2` before clarke and park so that this clipping doesn't happen as a motor is never going to enjoy a clipped sin wave.

If you agree with that then I believe the `voltage_limit > 0.5 * voltage_power_supply` doesn't make sense.  I'd also argue sending anything higher than `0.5 * voltage_power_supply` to `motor.move()` in voltage control mode doesn't make sense.  Same with `voltage_sensor_align`.

Might be easier to have a zoom call to discuss.

askuric, 2020-09-16:
Hey Owen. 
Your hack is a nice way to make the voltage limit more intuitive. But it would make the FOC algorithm a bit more complicated to read. Since it is an exact algorithm and this library is based on two modulation techniques (sinusoidal and space vecotor) I would certainly like to keep this part as close as it can be to "by the book". 

I agree that the voltage limit setting is not intuitive. What you are setting is the q component of the voltage seen by the motor. And depending on modulation technique that will mean different waveforms. Here is a really breef explanation of this in docs:
https://docs.simplefoc.com/foc_theory

Now I agree that this is not very well explained, and I somehow never found the time to do it well. Therefore maybe we can leave this issue open as long as it takes me to document it. :D 

If you watch carefully the waveform of the space vector and sinusoidal signal, you'll see that sine waves have a bit higher magnitude. That is due to different vector calculation and if you would control your motor with the same magnitude these two modulations will give you different power. Because motor will not see the same voltage (q component of the voltage).

Honestly I have had the voltage limit forced not to be higher than `voltage_power_supply/2` before but it seemed to mislead people even more. If you go though a bit earlier library versions you will see this feature. I am sure there are still some comments remaining from these times :D

And regarding the clipping, it is not 100% true that after `Uq > 0.5*voltage_power_supply` motor will not see any difference. This very much depends of the motor.

So to answer some of your questions:
1) Why do we center around `voltage_power_supply/1`?

Because the motor will see exactly the same voltage as if would center about some other value but this value ensures that all the switches of the transistors S1,S2,S3,S4,S5 and S6 are equally used.  And don't be fulled by the voltages values 2V, 3V and so on. The only voltages your motor ever sees are `0V` and `volatge_power_supply`. The only question is how much is which switch(mosfet/transistor) on.

2) Why not use `voltage_limit <= voltage_power_supply/2`?

This would make sense for `SinePWM` and not for `SpaceVectorPWM`. And additionally, the motor actually can see the difference (in power) even if the values are clipped.

3) Why not use magnitude (or half of the magnitude) and voltage set point and not q component of voltage seen by the motor?

This can be done, the equations will change and we would need to go through the math to verify that all the transformations are respected. But since this library is educational among other things, and since these algorithms are the core of it, I prefer to leave them as clean as possible and as close to the textbook ones as possible. 

We definitely can zoom for more in depth discussion.  Especially since I don't have time at this moment to go in depth with the explanation. :D

cousinitt, 2020-10-05:
Originally I started reading the source code here because it was the clearest source of information that I could find. I'm not a motor expert at all, but I want to learn. I noticed in the linked documentation posted by @askuric that the input to the inverse Park transform is u_q (I think the "q" means "quadrature"?). The article on Wikipedia on [vector control ](https://en.wikipedia.org/wiki/Vector_control_(motor))also has a direct component (d) that is linked with "field weakening". What sensor would we have to install in order to need the "d" component?

cousinitt, 2020-10-05:
I think @askuric already partly answered this in the [project documentation](https://docs.simplefoc.com/voltage_torque_control).

owennewo, 2021-01-02:
I'm the OP and am closing this issue as I'm happy with how this works

closed 2021-01-02 by owennewo