simplefoc/Arduino-FOC #22: can this lib support esc board ST B-G431B?

xlla, 2020-09-19:
I found this board integrate 3-current resistor and with F103 processor.
Can we use it to study simplefoc?

owennewo, 2020-09-19:
Hi @xlla . This board looks really interesting. I've not seen it before.
The main schematic here...
https://www.st.com/resource/en/schematic_pack/mb1419-g431cb-b01_schematic.pdf
Shows that it has two stm processors. A familiar stm32f103 (which i believe is the onboard stlink for programming) and a more exotic stm32g431. 
https://www.st.com/resource/en/product_presentation/microcontrollers_stm32g4_series_product_overview.pdf

Each motor phase has a l6387 which has high and low input pins. I.e this requires 6pin pwm.  SimpleFOC only supports 3pin pwm at the moment but 6 pin is definitely on the road map.

Current sensing is on the roadmap to.  This board does look like a candidate to experiment with and at $18 is great value.

Stm will expect people to use their ide and framework with this board (with their motor control apis) but i don't think there is anything stopping it being used to develop new features in simplefoc. I think the stmg32g431 is supported by platformio with Arduino framework.

If a few others want to do some hacking on this board, I'd probably join in the effort.

xlla, 2020-09-22:
@owennewo you are right!
I am looking for a cheap FOC esc board compare to expensive VESC board.
stmg32g431 has fpu and a lot of timer that can help speed up foc computations.

I prefer work with arduino framework to ST libs.

owennewo, 2020-09-27:
@xlla, it looks like @askuric is working on 6pwm which may mean this board might start to be usable with SimpleFOC
https://community.simplefoc.com/t/motor-driver-st-ihm08m1/131
This work won't unlock the current sensing capability of this board but is a step in the right direction.

owennewo, 2020-11-10:
@xlla. I've bought the b-g431b and have done a liitle work on getting it working.
The board wasn't officially supported by arduino,  I've created a PR to stm32duino to add this as a new board variant.
https://github.com/stm32duino/Arduino_Core_STM32/pull/1236
This basically would allow you to select the board in arduino ide.
I reckon that PR might take a few months (stm32duino are restructuring their repo and PRs are being blocked).
I've uploaded a couple of videos on youtube of what I've done so far with this board. @askuric is implementing 6pwm as we speak and I'm helping to ensure it also works on g431 chip.

Hopefully we'll have something working before the new year.

askuric, 2020-11-23:
Hey @xlla , 

The library version [v2.0](https://github.com/simplefoc/Arduino-FOC/releases/tag/v2.0) now supports the high/low side separate pwm setting so you should be able to run the code on this board as well.

closed 2020-12-14 by askuric