simplefoc/Arduino-FOC #23: [Schematic] Diode TBAT45s pin identification error

kingsjl, 2020-09-21:
Schematic diode pin identification error：
[image]

Correct pin sequence：
[image]

askuric, 2020-09-21:
Hey @kingsjl ,
You are absolutely right.
I'll update the schematic asap.

I will also create another git repo for the shield so that the versions are easier to track.

askuric, 2020-09-22:
Hey @kingsjl , 
The updated schematic and full source is now available in *Simple**FOC**Shield* dedicated repo:
https://github.com/simplefoc/Arduino-SimpleFOCShield

There you can find:
- schematics
- gerber files
- 3d models
- BOM file
- Pick and Place files
- Altium project

I still did not update docs website so I will leave this issue standing until that is done :D

kingsjl, 2020-09-22:
> Hey @kingsjl ,
> The updated schematic and full source is now available in _Simple**FOC**Shield_ dedicated repo:
> https://github.com/simplefoc/Arduino-SimpleFOCShield
>
> There you can find:
>
> * schematics [...]

Ok Thank you

askuric, 2020-10-08:
The docs have been updated and the git repo has been created for the board so this should no longed be a problem. :D

closed 2020-10-08 by askuric