simplefoc/Arduino-FOC #24: Linear 3 fase motor design

Juanduino, 2020-09-22:
Hi, Great work on this repo!

Im doing dev on a linear motor controller and have been looking at the SAMD51 chip supported by Arduino IDE. https://groups.google.com/forum/#!topic/openpnp/CnMdecdHXEY

Well. I just decided to change approach, and found your work. 

Regarding the current sense feature you want to implement. How about we develop this feature for this specific board? The SAMD51 is running 120mhz and it would be a shame not to give it something to work with. For a linear motor to have high precision, i suspect you need to sense the current on each fase, while measuring the hall sensors. I have not read up on the 3 fase controller you have been using, but i suppose one can use your lib with 3 dual switches? (Mosfets) Would it be an option to sense the total current for all 3 fases, and use that in the calculation to decrease complexity?

Juanduino, 2020-09-22:
This one goes up to 30 amp in a SO8 package.

https://www.mouser.dk/ProductDetail/Broadcom-Avago/ACHS-7123-000E?qs=w%2Fv1CP2dgqqiNyZEYkpD6A==

Juanduino, 2020-09-22:
[image]

Juanduino, 2020-09-22:
https://vimeo.com/267115723

Juanduino, 2020-09-22:
I see your code compiles for the feather M4. And it can be overclocked to 200 mhz. Probably with cooling :)

Juanduino, 2020-09-22:
Not yet fully routed...
[image]

owennewo, 2020-09-22:
It's good to hear the library works on samd51/feather m4.

I think what you are asking for here is to design a motor control board around samd51 and to add current sensing. There has been a bit of chat on current sensing on the [community](https://community.simplefoc.com/) website and one member (dgr) is designing the dagor controller that has current sensing hardware on it.

Juanduino, 2020-09-22:
https://www.youtube.com/watch?v=TLGWQfK-6DY&list=RDqORYO0atB6g&index=5

Yeb controller... Linear & rotational 360Ø

Juanduino, 2020-09-22:
[image]

Juanduino, 2020-09-22:
Ok i want spam your repo anymore. Just wanted you to see this: https://www.microchip.com/wwwAppNotes/AppNotes.aspx?appnote=en599773

Juanduino, 2020-09-24:
Ok im getting closer. 

I managed to fit in current amplifiers on each fase through 0.05ohm resistors. They output a voltage proportional to current through the resistor. 

Furthermore, i managed to get room for he HALL´s on the board. The magnets in the shaft of the linear motor is made of typical 6x3mm or 8x3 & 10x3mm magnets ... 3mm neodymium magnets are the most common on amazon and so on. The windings for the moving part of the motor will be wound around the magnets and therefore it will use the entire pull of the magnet. The magnetic field of the winding's will move in the direction of the movement and will not be as sensitive to distance between magnets and windings. 

I im now integrating the TMC2209 stepper driver, since the use case for this specific dev. is z axis and rotation on a OpenPnP machine (using a linear motor for z and a nema8 stepper for rotation). The linear motor concept is based on the same principle (3 fase) as regular BLDC, so im hoping to make this work using this awesome library. OpenPnP can use subdrivers for motion. This means you can have one controller connected through USB for each nozzle on the machine (maybe 4) and save on cables and overall weight to move around. Having the stepper driver close to the nema8 is god because of the short cable distance, for keeping resistance in the windings down. 

[image]
[image]

Juanduino, 2020-09-24:
[image]

https://www.youtube.com/watch?v=7fD5hz8bHtU&list=RDorbCZooMvQs&index=2

owennewo, 2021-01-02:
@Juanduino - is this issue still active.  I was going to close but thought I'd check first.  

The issue feels more like a question/collaboration so might be better answered on https://community.simplefoc.com forum

closed 2021-05-13 by askuric