simplefoc/Arduino-FOC #25: Controlling Low_side High_side dual switches

Juanduino, 2020-09-25:
Hi

I have been reading up on the ATSAMD51 MCu since i have decided to try your lib. on this chip.

The way you have implemented the pwm driver makes sense for the motor driver ic on your shield, but if you want to control logic level mosfets (single or dual) you have to switch 6 outputs. High side and low side.

The SAMD51 can sync. pwm outputs/channels for switching/motor driving applications. This insures that the two synchronized outputs is not eg. high or low at the same time.  To set this up, you have to access the registers somehow during startup, either directly or using Arduino syntax. It is a bit confusing, since i just started reading up on the datasheet.

I think i somewhere red, that this is a todo?

What are your thoughts on this?

askuric, 2020-09-27:
Hi @Juanduino 
Your question is spot on, I have had this discussion with a lot of people now and I will definitely try to include it to the next release. At least for the STM32 boards.

I need to do some extensive testing and find a way to integrate it meaningfully into the library. There will be a lot of hardware specific code to write. :D

I've created milestone for this issue, so I'll keep you updated with the progress.

Juanduino, 2020-10-01:
Hi @askuric 

Yes, it’s quite the task. Maybe there is inspiration in the VESC repo.

Like I mentioned, I have been doing work on a linear 3 phase controller. Here is the work in progress. https://github.com/Juanduino/THE-DIMMER_M4SDRV_V1

In a few weeks time I will order prototypes and would be glad to send you out one. On that occasion I will make a small group purchase of PCB and components.

askuric, 2020-11-23:
Hey @Juanduino 

I hope you are doing well and that your cool project is advancing!

The simplefoc library now supports high/low side pwm control, you should check it out: https://docs.simplefoc.com/drivers_config

This mode is supported for atmega328, stm32 and esp32 boards.

Cheers,
Antun

closed 2020-12-14 by askuric