simplefoc/Arduino-FOC #33: Incorrect bit mask created in MagneticSensorSPI::read

GIT-JoshDS, 2020-10-23:
I noticed I was getting lots of "noise" when using an AN5048A mag angle sensor. The reported sensor angle was constantly flickering between two values. The two values changed as expected when rotating the sensor, but always oscillated between two values with a very constant offset of approximately 6.3 radians.

After chasing the source of this flickering around I realised that it was probably the parity bit flickering on and off with one of the LSB's in the angle reading. After looking through the source, I found this line in the MagneticSensorSPI file. 

`const static word data_mask = ~(0 >> (16 - bit_resolution));`

The AS5048 is a 14 bit sensor so this works out to ~(0 >> 2). Right-shifting 0b0 by 2 equals 0b0, ones-comp of 0b0 is -0 or 0b1111111111111111.

After changing this line to...

`const static word data_mask = 0b11111111111111;`

The flickering stopped and the reported angle behaved as expected, smoothly following the rotation of the sensor. There is still some noise in the line because of the minor bits flipping back and forth but it's barely noticeable.

Either of the lines below should work as replacement.

`const static word data_mask = 0xFFFF >> (16 - bit_resolution);`

`const static word data_mask = pow(2, bit_resolution) - 1;`

Thanks for your work on the library. Its been really helpful.

Cheers,
Josh

owennewo, 2020-10-24:
Looks like we spotted this at pretty much the same time!

Does this look good to you?
https://github.com/simplefoc/Arduino-FOC/pull/32

Apologies - I introduced this bug when I was refactoring.

BTW can I trouble you for your configuration of your sensor? e.g. do you use this:
```
// this is for as5047u
MagneticSensorSPI sensor = MagneticSensorSPI(10, 14, 0x3FFF);
```
I'm trying to simplify the way that these are setup so that you don't need to go digging in the datasheet.

GIT-JoshDS, 2020-10-24:
Yeah, that looks good.

I'd been digging through stuff for a while trying to work out a couple issues with my set-up and saw the line, in my brain-fog I mis-remembered the tilde operator working in some weird way.

I think its actually the order of operations throwing it off though now that I look at it. The outer set of brackets means the zero gets right shifted before being inverted. Without them it probably works fine.

Though as you mention you might still have issues with the zero being interpreted as an int and doing funny things on different hardware.

I use the MagSensSPI(SS_pin, bit_res, angl_reg) option. I did add a config option for the AS5048 to the top of the .h file with the others but havent tried it yet. And the default 10, 14, 0x3FFF work.

I'm trying to make a linear inverted pendulum using a gimble motor with AS5048 and a second AS5048 for the pendulum angle and a Nucleo L486RG. Bridged two pins with solder to run them in 3v mode for the Nucleo but Im getting weird issues trying to run the two sensors through the SimpleFOC lib. Spent enough time on it today though, try again tomorrow...

Cheers

closed 2020-10-24 by GIT-JoshDS