simplefoc/Arduino-FOC #34: About SPI encoder

zhuangfengzi, 2020-11-07:
The accuracy of the tle5012b encoder is 16 bit or 15 bit, and the performance is very good! His SPI was originally 3 wires, and the official added 4-wire library, which is convenient to connect! Can add this encoder may give me some advice on code transformation! esteem it a favor!

https://github.com/Infineon/TLE5012-Magnetic-Angle-Sensor

zhuangfengzi, 2020-11-07:
Reading angle test code works well! As follows:

#include "Tle5012b_4wire.h"

Tle5012b_4wire Tle5012MagneticAngleSensor;
errorTypes checkError = NO_ERROR;

void setup() {
  delay(2000);
  Serial.begin(115200);
  checkError = Tle5012MagneticAngleSensor.begin();
  Serial.print("checkError: ");
  Serial.println(checkError,HEX);
  delay(1000);
  Serial.println("Init done");
}

void loop() {
  double d = 0.0;
  double r = 0.0;
  double s = 0.0;
  double t = 0;
  int16_t b = 0;

  Tle5012MagneticAngleSensor.getAngleValue(d);
  Tle5012MagneticAngleSensor.getNumRevolutions(b);
  Tle5012MagneticAngleSensor.getAngleSpeed(s);
  Tle5012MagneticAngleSensor.getAngleRange(r);
  Tle5012MagneticAngleSensor.getTemperature(t);

  Serial.print("Temperature:"); Serial.print(t);
  Serial.print("°C\tangle:");   Serial.print(d);
  Serial.print("°\trange:");    Serial.print(r);
  Serial.print("\tspeed:");	    Serial.print(s);
  Serial.print("\t Revolution:");Serial.println(b);

  delay(10);
}

owennewo, 2020-11-08:
This sensor is somewhat different to the magnetic spi sensors we currently support. E.g returns 4 bytes not 2. 

If you want to try to add support for this sensor then I'd suggest starting with the magneticsensorspi.cpp and changing it to meet your sensor.

zhuangfengzi, 2020-11-09:
@Owennew o, thank you!

Tle5012b was originally a three line SPI, but it was changed to a 4-line SPI, but the SPI can work. It is only an official library, but I don't know how to transplant the official library into simple FOC, or after integration,

This can happen:

MagneticSensorSPIConfig_ s AS5147_ SPI = {

.spi_ mode = SPI_ MODE1,

.clock_ speed = 1000000,

.bit_ resolution = 14,

.angle_ register = 0xCFFF,

.data_ start_ bit = 13,

.command_ Rw_ bit = 14,

.command_ parity_ bit = 15

}

/** Typical configuration for the 14bit MonolithicPower MA730 magnetic sensor over SPI interface */

MagneticSensorSPIConfig_ s MA730_ SPI = {

.spi_ mode = SPI_ MODE0,

.clock_ speed = 1000000,

.bit_ resolution = 14,

.angle_ register = 0x0000,

.data_ start_ bit = 15,

.command_ Rw_ bit = 0, // not required

.command_ parity_ bit = 0 // parity not implemented

}

MagneticSensorSPIConfig_ s Tle5012b_ SPI = {

.spi_ mode = SPI_ MODE0,

.clock_ speed = 1000000,

.bit_ resolution = 15,

.angle_ register = 0x7FFF,///???????????

.data_ start_ bit = 15,

.command_ Rw_ bit = 0, // not required

.command_ parity_ bit = 0 // parity not implemented

}

.........

You can also use the code magneticsensorspi sensor = magneticsensorspi (16, 14, 0x3fff); I'm a novice, so it's hard!

zhuangfengzi, 2020-11-09:
@owennewo
There is this column in the official document. I am looking at it. I will help you to see it under the guidance!
https://docs.simplefoc.com/sensor_support

owennewo, 2020-11-10:
Is there any reason why you want to use this particular sensor? It is expensive (£35) and currently not supported. 
You'll need to rewrite most of the class, I can't write this for you as I don't have the sensor. 
Can you code c/c++? How determined are you to complete this?

zhuangfengzi, 2020-11-12:
It's half cheaper than as5047p as5147p, so use this test! If there is no way, only change the encoder! My C + + is very bad, just a fan! @Owennew o, thank you very much!

runger1101001, 2021-05-26:
Dear @zhuangfengzi ,

I have some TLE5012 on my desk and will try to get them supported in SimpleFOC soon. Some other users have recently done some work on this as well (https://github.com/xerootg/btt-s42b-simplefoc/blob/master/src/TLE5012b.cpp), but it is hardware specific code.

You are right that the chip is less expensive most of the others, just the evaluation boards are more expensive :-)
For this reason alone I think it is interesting to support it.