simplefoc/Arduino-FOC #40: Annoying sound of the motor~

HansAItec, 2020-12-14:
I try to learn the BLDC control with Arduino Mega2560,  SimpleFoc and Ipower 5208 motor.
But the motor have some electrical sounds just like tinnitus, maybe it's because of th PWM frequency or something...
So is there anyone who can help me with this problem?
How can i change the PWM frequency?
tks!

askuric, 2020-12-14:
Hey Hans, we've recently made a support for Mega2560. And it is still in the dev branch.
We will definitely merge it to the new release in a few days.
Till then, you can use the dev branch code :D

askuric, 2020-12-22:
Hey @HansAItec 
The new release is out [v2.0.2](https://github.com/simplefoc/Arduino-FOC/releases/tag/v2.0.2) and now the library supports the Arduino Mega2560. So I'm closing this issue!

closed 2020-12-22 by askuric