simplefoc/Arduino-FOC #41: _constrain modify Ua Ub Uc

binsruby, 2020-12-15:
I use 6pwm driver ,i found Ua Ub Uc  voltages is modified to positive when one is negative in this function below
[image]

will this have effect on FOC?

askuric, 2020-12-15:
Function setPWM receives only positive values. Simplefoc will never send negative values to this function. 

This code limits the maximal voltage we want to set, and this is the vital part of the code, so I'm pretty sure it works well.

Was this your question?

binsruby, 2020-12-15:
> Function setPWM receives only positive values. Simplefoc will never send negative values to this function.
> 
> This code limits the maximal voltage we want to set, and this is the vital part of the code, so I'm pretty sure it works well.
> 
> Was this your question?

this function just calculate duty cycle.  if Ua = -4.0 volt， then after _constrain,  Ua=0，dc_a = 0,   it means this pwmA_h pin side  has no effect on field control,   but in fact  calculate value is -4.0 volt.   is thart right?

askuric, 2020-12-15:
Again, you cannot set negative values. For each of the phases you can set the voltage in the range from 0 to the power supply voltage ( and this constrain limits it to `voltage_limit`.

So to answer your question, if you try to set negative voltage to one oft he phases, the setPwm will set 0. But please bare in mind that this is not a choice, this is the hardware limitation. Any BLDC driver you use no matter if it has 3 or 6 PWM signals cannot set negative voltages.

Basically this function calculates the values of duty cycle which correspond to the voltages you want to set. And the minimum duty cycle you can set is 0 which is basically 0 voltage. The maximum duty cycle you can set is 1 which is power supply voltage.

binsruby, 2020-12-16:
ok，thank U

askuric, 2020-12-16:
Hey @binsruby ,
I hope you've gotten your answer. 

If you wish to read more deeply about what does the library actually do and what kinds of waveworms does it set to the driver please check the digging deeper page:
https://docs.simplefoc.com/foc_theory

In the mean time, since I did not spend much time documenting properly the foc and drivers I will leave this issue open until I document better the library foundation.

Antun

closed 2021-01-06 by binsruby