simplefoc/Arduino-FOC #45: PWM Frequency doesn

polymetric, 2021-01-01:


owennewo, 2021-01-02:
We really welcome people trying new chips!!

The due uses a 32bit SAM3X8E chip which has a `analogWriteFrequency()` method.

The SA3X8E is a type of SAM3XA chip so perhaps targetting `#define _SAM3XA_` for chip detection is appropriate.

I can't test (I have no DUE) but I'd try copy/rename the file `teensy_mcu.cpp` (link below) to `samxa_mcu.cpp` and then changing the `#define` to `#if defined(_SAM3XA_)` logic in new file and in `generic_mcu.cpp`

https://github.com/simplefoc/Arduino-FOC/blob/master/src/drivers/hardware_specific/teensy_mcu.cpp

Putting it another way I suspect that our hardware specific logic for teensy would work for due - but probably worth breaking them out as they may diverge.

askuric, 2021-01-25:
Hey guys, 
This is a great issue, but we will need some help here I think. 
I do not own a due  and we will need some volunteers to provide us with some examples of working codes samples. Otherwise it will take some time until we resolve it and it is a shame a bit :D

askuric, 2021-01-29:
Arduino DUE should is supported from this moment. The support is currently in the dev branch and it will pass in the next release. It supports all drivers and all the features except 6PWM mode. If I will have enough time I'll try to squeeze that as well into the next release.  Here is the link for current version of the library in dev:
https://github.com/simplefoc/Arduino-FOC/tree/dev

The support enables combining the pwm and timer pins at the same time and for now we support only the pwm pins supported with the Arduino implementation. Let us know if you are interested in having fuller support!