simplefoc/Arduino-FOC #46: motor.command() and CRLF

owennewo, 2021-01-02:
This confused me - so I'm raising an issue for discussion.

`motor.command(String)` doesn't work well with Serial set to eol=CRLF (i.e. `\r\n`)

With CRLF all commands are 'sets' e.g. `P` will set the P term to zero.

A fix for this is to either configure your serial terminal e.g. platformio (which defaults to CRLF) you could set the following in you platformio.ini
```
monitor_flags =
  --eol
    LF
```

If we wanted to be more 'tolerant' we could change the code e.g. to be:
```
  char GET = user_command.charAt(1) == '\n' || user_command.charAt(1) == '\r';
```

askuric, 2021-01-27:
https://community.simplefoc.com/t/node-oriented-communication-interface-for-the-simplefoc/466/14

askuric, 2021-05-10:
You're right. If you have time go for it. :)

askuric, 2021-05-10:
https://community.simplefoc.com/t/node-oriented-communication-interface-for-the-simplefoc/466/15

closed 2021-05-13 by askuric