simplefoc/Arduino-FOC #47: switch off phases completely

m84u, 2021-01-04:
To 'deactivate' the driver/motor, to save batterie energy, we should have the possibility to do it by using the api.

askuric, 2021-01-27:
We'll make sure to implement it with the next release. Here is the full discussion: https://community.simplefoc.com/t/high-current-when-motor-isnt-spinning/414/7

askuric, 2021-09-16:
https://community.simplefoc.com/t/b-g431b-esc1-freewheeling/1183/4

runger1101001, 2021-09-16:
Looking at the code, I think driver->setPwm(0, 0, 0)
doesn't wind up turning the FETs completely off in all cases, for example in Software 6-PWM for STM32.
It's because of the way we do the contain with the dead-time

Iurii-Dolgorukii, 2022-11-26:
Hello friends. It's been almost two years. Has the function of disabling all phases been added?))) I use 6 pwm and esp32. I wanted to use it for a bicycle. But when I want to roll, the mosfets of the lower side open and the motor windings are switched, which is why I instantly lose speed

runger1101001, 2022-11-29:
Hi @Iurii-Dolgorukii , we've actually been working on it, and there is a PR here waiting to be tested and merged:

https://github.com/simplefoc/Arduino-FOC/pull/231

So you could try this code, and help us to test it, if you want, or else if you are patient for a little longer, we will try to merge this soon.

askuric, 2022-12-22:
The code is in the `dev` branch now. 
Its a combination of pull requests: #231 and #213 
For now switching phases off works for stm32 in 6pwm mode both in hardware and softrware 6pwm. 
It also works for atmega328 and atmega2560.

Iurii-Dolgorukii, 2022-12-26:
@askuric 
Hi. For stm32 this works when calling motor.disable(); ?

askuric, 2022-12-26:
Hey @Iurii-Dolgorukii,
Yes this will work with the library version from the dev branch.

runger1101001, 2023-03-21:
This is now merged to the main branch, and released in version 2.3.0 of the library :-)

closed 2023-03-21 by runger1101001