simplefoc/Arduino-FOC #49: Use previous value of getAngle() in getVelocity()

runger1101001, 2021-01-26:
At the moment the getVelocity() function of Sensor gets called from move(), and in turn calls the getAngle() function. getAngle() is also called from loopFOC().
Since reading sensors takes time and is the limiting factor for the FOC-iteration rate on fast MCUs, and a significant contributor on slow MCUs, it could be a good idea to reduce the number of sensor reads by 1 per iteration, by remembering the last value in getAngle() and using this in getVelocity(). 

One way to do it might be to change the signature of getAngle() to getAngle(bool refresh = true); 
I'll supply a pull-request for this if I can.

I think this is especially important for slow MCUs like the ATMega328P trying to run multiple motors (like on the older gimbal controllers).

See also Forum post: https://community.simplefoc.com/t/atmega328pb-for-sfoc-control-with-two-as5600-i2c/371/12

runger1101001, 2021-05-04:
The AS5048A driver [here](https://github.com/simplefoc/Ardunio-FOC-drivers/tree/master/src/encoders/as5048a) implements this behaviour. I'm using it everywhere now, with no noticeable difference in the performance of my simple test setups.

askuric, 2021-05-05:
Great!
Do you think it is a good idea to keep the getAngle timestamp not do it only if the calls of getAngle and getVelocity are too close (some arbutus time)

To enable the getVelocity stand alone call and to enable more up to date velocities if the somehow angle is called some time ago.

askuric, 2021-07-20:
PR #91 has pretty much resolved this issue for now I think

closed 2021-10-31 by askuric