simplefoc/Arduino-FOC #63: Could you add support for LGT8F328P 32MHz similar to atmega328p but cheaper

z13660, 2021-03-30:


runger1101001, 2021-03-30:
Board manager URLs: https://github.com/dbuezas/lgt8fx
Datasheets: https://github.com/watterott/LGT8F328P-Testing

We can take a look, but TBH supporting a chip where the test boards have to be ordered custom-made is probably not a big win, compared to say, supporting the Raspi Pico, nRF52840 or other, more widely used and well-documented chips...

Is this chip widely used? Are there ready-to-order boards with this chip commercially available? Would you have an AliExpress Link or something you can share?

In the meantime, if you are keen you could take a look at the source code here: https://github.com/simplefoc/Arduino-FOC/blob/master/src/drivers/hardware_specific/atmega328_mcu.cpp
That's the ATMega support. If the chip is really compatible, it should be mainly a matter of changing some timings... if you have such a chip you can test with, maybe you want to take a look at implementing it and send us a pull request?

z13660, 2021-03-31:
Thanks for your reply. 
test board are available on(arduino nano like)
https://m.aliexpress.com/item/1005001693711452.html
The chip has a 12bit adc maybe better at current sensing. I will order and try it.

wimalopaan, 2021-09-14:
Are there any experiences already with these boards?