simplefoc/Arduino-FOC #65: Will this library support stm32f405？ will this lib support the odrive board？

Wenuo, 2021-04-07:


runger1101001, 2021-04-07:
STM32F405 is supported.
ODrive is currently not supported, but I believe some people are working on it. TBH, ODrive has its own FOC implementation, and is "ready to run" with the ODrive Firmware, so making it SimpleFOC compatible isn't a clear win, IMHO.

May I ask why you'd like to use SimpleFOC with ODrive? Is there something specific ODrive does not do for you with its normal software?

Wenuo, 2021-04-07:
Thank you for your reply！and i made the odrive board with myself pcb layout just now ,but the Odrive Fireware not working .That's why I asked this question

owennewo, 2021-04-09:
I've got SimpleFOC running on VESC which is stm32f405 based.  The trickiest part for getting a new stm board working is HSE/HSI clock config.  VESC uses 8Mhz ocillator - if odrive uses 8 drive then the clock config should be the same.

I intend to package the VESC variant files and submit them to stm32duino (so that the board can easily be used with arduino).  In the mean time I've got them on a branch here:
https://github.com/owennewo/Arduino_Core_STM32/tree/vesc_4_12/variants/VESC_4_12
And if you use platformio (like me) then the board can be registered with:
https://github.com/owennewo/platform-ststm32/blob/vesc_4_12/boards/vesc_4_12.json

If all this seems overly complex then I'd start by copy/pasting function SystemClock_Config into the top of your arduino ino and setting a build flag for the oscillator i.e. -D HSE_VALUE=8000000

Wenuo, 2021-05-03:
Thank you very much, my friend. Do you just copy a function that SystemClock_Config and define it HSE_VALUE=8000000 ?

Wenuo, 2021-05-04:
Just like this？ @ owennewo

#define HSE_VALUE 12000000U
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSI|RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.LSIState = RCC_LSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 4;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

    /**Initializes the CPU, AHB and APB busses clocks 
    */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;

  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
}

SoftwareSerial mySerial(PA0,PA1); // RX= PA0
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  //mySerial.begin(9600);
  pinMode(PB3, OUTPUT);
  digitalWrite(PB3,LOW);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PB3, HIGH);   // turn the LED on (HIGH is the voltage level)                       // wait for a second
  delay(10000);
  digitalWrite(PB3,LOW);
  delay(500);
  //mySerial.print("Hello world!\n");
  //delay(2000);
}

Wenuo, 2021-05-04:
@owennewo just like this？Simply change HSE_VALUE = 8000000U? Looking forward to your reply！
#define HSE_VALUE 12000000U
extern "C" void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_LSI|RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.LSIState = RCC_LSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLM = 4;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

    /**Initializes the CPU, AHB and APB busses clocks 
    */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;

  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK)
  {
    Error_Handler();
  }
}

SoftwareSerial mySerial(PA0,PA1); // RX= PA0
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  //mySerial.begin(9600);
  pinMode(PB3, OUTPUT);
  digitalWrite(PB3,LOW);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PB3, HIGH);   // turn the LED on (HIGH is the voltage level)                       // wait for a second
  delay(10000);
  digitalWrite(PB3,LOW);
  delay(500);
  //mySerial.print("Hello world!\n");
  //delay(2000);
}

closed 2021-05-05 by Wenuo