simplefoc/Arduino-FOC #70: False Clarke Transform in FOC

Nemantor, 2021-05-01:
While trying to implement the foc algorithm for a delta wound linear motor, I stumbled upon a little mistake in clarke transform which resulted in unstable current control loop. 

`        i_alpha = 0.6666667*(current.a - (current.b - current.c));    
        i_beta = _2_SQRT3 *( current.b  - current.c );`

If am not wrong It is supposed to be like this

`    i_alpha = 0.6666667*(current.a - current.b / 2 - current.c / 2);    
    i_beta = 0.57735 *( current.b  - current.c );
`

askuric, 2021-05-01:
You're absolutely right!
I've tested the code mostly for 2 adc signals and since I'm did not have a setup with 3 adcs.
I'll update the code for the next release, unless you wanna make a pull request?
Thanks!

Nemantor, 2021-05-01:
I would really like to.  I also made some modifications in order to find currents on a delta wound motor from two current sensors.

Polyphe, 2021-07-12:
Hi Nemantor,
Amazing ! I had never done the difference between a star winding and delta winding in FOC in motion control...
Why to divide by 3 instead to divide by square root of 3 ?
So from your formulas without (* -1) => 
I_alpha = -(I_b - I_a) / 3.0f;
I_beta = _1_SQRT3 * (-(I_b - I_a) / 3.0f) + _2_SQRT3 * (-I_a);

Is it correct ?

Thank you in advance for your answer.