simplefoc/Arduino-FOC #71: Strange SVPWM waveform using atmega328p

z13660, 2021-05-01:
motor running at open_loop_velocity_example.ino,motor can spin like normal,but when I connect to the oscilloscope, it is not a standard svpwm waveform I tried to change several atmega328p s,but the same.
https://user-images.githubusercontent.com/23288802/116771035-d028d100-aa7a-11eb-9048-c2614f32f1c7.mp4

askuric, 2021-05-01:
Hey @z13660 
It doesn't look like space vector owl indeed. Did you enable it in code?
`motor.foc_modulation =FOCmodulationTtype::SpaceVectorPWM`
By default it's sinusoidal pwm, and in the scope it looks like sine owl to me :)

askuric, 2021-05-07:
Hey @z13660,
have you had some time to investigate more the pwm signals?

z13660, 2021-05-07:
I will test it these days. The oscilloscope is at school. Please wait for a little longer. (^_^)

z13660, 2021-05-09:
OK I think I figure it out. Is it the timer problem? Changed the pin BLDCDriver3PWM(9, 10, 5, 8); and it work perfectly.

https://user-images.githubusercontent.com/23288802/117557894-6deb5400-b0aa-11eb-94da-9be0eec21149.mp4

askuric, 2021-05-09:
Hey @z13660 ,
Can you elaborate a bit more about what do you mean that it is not a good waveform?

z13660, 2021-05-09:
[image]
[image]

z13660, 2021-05-09:
I don`t know if the phase shift will cause any problems. e.g. like causing the motor spinning less smooth.

askuric, 2021-05-09:
Hey @z13660,

Yes, I see what you mean. This is true, and if you look more closely you'll see that even in the image above pin3 is not perfectly aligned. This is due to the fact that the pins do not belong to the same timer. So far in the library, at least for arduino we did not do the timer synchonisation, mostly because arduinos are very limited in their capabilities and a lot of standard features do not even exist for them.

The shifted phase will not do much harm, at least if the shift is not too severe and if the pwm frequency is high. So that is why we did no see many complaints so far. But you're right that we should dig into this more carefully. 

I've tested a quick code that was posted in this site and it worked for me:
http://www.openmusiclabs.com/learning/digital/synchronizing-timers/index.html

So basically add this code right after your `driver.init()`:
```cpp
  // use only one of the following 3 lines
GTCCR = (1<<TSM)|(1<<PSRASY)|(1<<PSRSYNC); // halt all timers

// place all timer setup code here
// do not do any timer setup before this section

// set all timers to the same value
TCNT0 = 0; // set timer0 to 0
TCNT1H = 0; // set timer1 high byte to 0
TCNT1L = 0; // set timer1 low byte to 0
TCNT2 = 0; // set timer2 to 0

GTCCR = 0; // release all timers
```

This code should sync all the timers.

z13660, 2021-05-09:
Hey! @askuric

z13660, 2021-05-09:
Hey @askuric,
Thanks for your codes. (^_^)

askuric, 2021-05-09:
I'll leqve this issue opened for a bit. I'd like to get to the bottom of this. :D

askuric, 2022-12-22:
Included in the dev branch, and will be in the next release

closed 2022-12-22 by askuric