simplefoc/Arduino-FOC #74: Incremental encoder AS5047P using the abi interface wrong reading

Wenuo, 2021-05-05:
Hello，

I tried to run simplefoc on the odrive board,Just use the AS5047P AB interface and ignore the I interface，It runs normally. But I tried to use the I interface, The Encoder read angle error.At the index signal, the signal begins to beat. When I run position closed loop, the encoder calibration is not normal. 

I really don't know how to solve this problem. The Encoder ppr is 1000. Could you tell me how to solve?
Thank You for the Answers :)

askuric, 2021-05-05:
Hey @Wenuo !

Can you dump your monitoring output of the motor init?

Also are you sure you have the index enabled?
I worked a bit with as5047 sensors and I am not sure if the index pin was enabled by default. 

You're using an ODrive board, you've managed to get it to work?

Wenuo, 2021-05-05:
Thank you for your reply！
@askuric 
odrive runs simplefoc perfectly！(just use abi interface except i interface，Because I only have this encoder).
I use the 
Encoder encoder = Encoder(PB4, PB5, 1000, PC9);
void doIndex(){encoder.handleIndex();}
encoder.enableInterrupts(doA, doB, doIndex);
When calibrating the encoder, the motor jumps in the second half circle.

askuric, 2021-05-05:
Hey @Wenuo,

I think that your Intex pin is not enabled and that is why you are having the issues.
Here is the AS5047P datasheet: https://ams.com/documents/20143/36005/AS5047P_DS000324_2-00.pdf
Check the page 18.

Is it possible that you your ABI_UVW is set to 1 and that you have some PWM signal on you I pin instead of the index pin?
Can you try to read the I pin on its own and see if it turns on only one time per revolution?

You could do something like
```cpp
void doIndex(){
encoder.handleIndex();
Serial.println("index interrupt");
}
```

Wenuo, 2021-05-06:
Hey @askuric ,
I decided I don’t need to index the pins anymore，and i have been finished double motor position open loop  and Speed ​​open loop,In the future, I will continue to complete the torque closed loop of the dual motor and position close loop and soon on.Can I upload my code samples to here, and contribute to simplefoc.
Looking forward to your reply！Thanks! : )

askuric, 2021-05-07:
Hey @Wenuo,

Did you manage to get to the bottom of your problem, was it the AS5047 sensor not configured issue?

What kinds of code samples do you wish to upload?
Do you think it makes sense to push it to this issue?

If not I'd suggest you to go to our community forum and make a short post and upload the necessary code there. The forum posts are a bit easier to read and follow and they tend to be a good starting point for a lot of people :D

Let me know what do you think!

closed 2021-07-20 by askuric