simplefoc/Arduino-FOC #76: ESP32 mcpwm_get_frequency: division by zero

jkirsons, 2021-05-10:
**Problem**
This line causes a division by zero when running the code on an ESP32:
https://github.com/simplefoc/Arduino-FOC/blob/ee2dfdeee62bc28fdee5820fb1d26a0af4dc80c9/src/drivers/hardware_specific/esp32_mcu.cpp#L105

**Cause**
The ESP32 Arduino Release 1.0.6 uses ESP-IDF 3.3.5
In ESP-IDF 3.3 in mcpwm.c, function mcpwm_set_frequency() called by mcpwm_init() does a division by frequency:
https://github.com/espressif/esp-idf/blob/7c86027531ebffe937c9d9d1080cd433eb993f35/components/driver/mcpwm.c#L143

**Possible Resolution**
To correct this, can we set pwm_config.frequency before calling mcpwm_init() ?  Eg add this immediately before:
`  pwm_config.frequency = pwm_frequency;`

askuric, 2021-05-13:
Ah, nice!
So the mcpwm_init() now sets the pwm frequency?
This would remove the need for the fancy calculation of prescalers and the periods.

I'll look into it and if it works it will simplify the code quiet a bit!

askuric, 2021-05-13:
Hey @jkirsons 

I've tested the `mcpwm_init()` frequency setting feature and the good news is that it works.
The bad news is that they used a fixed prescale and therefore lose a lot of precision for higher frequencies.
For example if you set 30kHz, `mcpwm_init()` will set 33kHz
If you set 40kHz, it will set 45kHz

This is not a huge deal but for now I'd leave our implementatio as is, regarding the frequency setting but hwever we'll add the line as you suggested it to avoid the  zero division errors:
`pwm_config.frequency = 2*pwm_frequency;`

jkirsons, 2021-05-13:
Thanks! It may save others some time figuring out what is wrong.

mcpwm looks like it changes a bit in IDF 4 versions, so there may be further changes when new ESP32 Arduino versions are released.

closed 2021-05-13 by askuric