simplefoc/Arduino-FOC #81: Discuss: feed forward controller implementation

runger1101001, 2021-05-26:
Discussion in forum here: https://community.simplefoc.com/t/b-g431b-esc1-hoverboard-motor-not-working-with-closed-loop-modes/898/18?u=runger

Do we want to add a feed-forward controller? discuss and schedule implementation if people agree it's worth it.

askuric, 2021-06-06:
It is an interesting question!
But I think there are two questions here;
1. do we include feedforward in our control loops
2.  do we add a new way to control the motor where user can set a sequence of 
`target_angle target_velocity target_torque`  or `target_velocity target_torque`

#### do we include feedforward in our control loops
For the first one, I honestly think we should not do it. It is too complicated for now and it would involve a lot of apriori knowledge of the trajectory the user wants to execute.  We can discuss this more in detail when we meet the next time. But feedforward term only makes sense if we know the trajectory (angle or velocity trajectory) in advance and we wish to force certain profile of velocity and torque for it. This is a great feature but for now out of scope of the simplefoc.

Now, what we could maybe integrate is a way for users to design their own controllers and link them to the simplefoc. 
```cpp
Controller my_new_controller();

 my_new_controller::do(target, measurement){
 // do some fancy stuff
 // return voltage/current command  
}

void setup(){

  motor.linkController(&my_new_controller());
}
```

And so that the guys can create their own smarter and more suitable PIDs and control loops.

#### do we add a new way to control the motor where user can set a sequence angle+velocity+torque
This one I'd say absolutely.

At the moment I see it as an extension of the commander interface. Where the user can set the in adition to the target value, multiple target values.

So instead of setting:
```sh
M10.5              # 10.5 rad
```
the user could do:
```sh
M10.5 100 3    # 10.5 rad, 100 rad/s, 3Volts
```
So then the simplefoc would automatically update the `voltage_limit` ,`current_limit` and `velocity_limit`.

In the velocity mode the use can set: 
- voltage mode: `target` `voltage_limit`
- current mode   `target` `current_limit` `voltage_limit`

Angle mode:
- voltage mode: `target` `velocity_limit` `voltage_limit`
- current mode   `target` `velocity_limit` `current_limit` `voltage_limit`

Torque mode
- voltage mode: `target` 
- current mode   `target` `voltage_limit`

What do you think about that?

askuric, 2021-11-07:
In the dev branch at the moment the commander was extended wit the new target setting interface. It enables setting the target and limiting variables at once. User can set target angle, velocity and torque at the same time. The values are sent separated by a space.

Example. P2.34 70 2
`P` is the user defined command, `2.34` is the target angle `70` is the target 
velocity and `2` is the desired max current/voltage(depending on the torque mode).

Depending of the motion control mode one is using, the commander will expect different number of inputs:
  - torque   : torque (ex. P2.5)
  - velocity : velocity torque (ex.P10 2.5) 
  - angle    : angle velocity torque (ex.P3.5 10 2.5)
  
  To use this interface in the code you just add it to the commander:
```
void doTarget(char* cmd) { command.motor(&motor, cmd); }
  .....
void setup(){
  .....
  command.add('T', doTarget, "motion control interface");
  .....
}
```

askuric, 2022-09-04:
To think about: https://community.simplefoc.com/t/motion-planner-code/2424/2

closed 2026-02-18 by askuric