simplefoc/Arduino-FOC #82: MagneticSensorSPI and STM32

runger1101001, 2021-05-26:
As reported here:
https://community.simplefoc.com/t/sensor-as5047p-spi-reads-0-angle-with-stm32f103-aka-blue-pill/886/18

but also by other users, I think. It explains the difficulties some people (including myself) were having with magnetic sensors and STM32.

Here is my summary from that post:

Hey, I did a quick test setup:

image
image
1000×692 191 KB
SimpleFOC 2.1.1 master branch (current normal version of the library)
BluePill STM32F103C8
AS5047P sensor
PlatformIO, STM32 support v13.0
Motor not connected, just the sensor
I can report the following:
It actually works out of the box, with no modifications to MagneticSensorSPI.cpp.
The resulting speed is too fast for my crappy logic analyser, the oscilloscope shows the clock line is at 9MHz:
[image]
That makes sense since we set the pre-scaler to 8, and 72MHz system clock / 8 = 9MHz SPI clock in this case.

Surprisingly, this is working fine even via the SPI hub and all the cables:
[image]
With my suggested modifications to MagneticSensorSPI, the picture changes to:
[image]
So now we don’t set the pre-scaler, but rather allow beginTransaction() to apply the clock_speed as specified in the settings object.
So we get 1MHz SPI clock, as specified in the settings.

To me this confirms the following:

the standard settings kind of work with STM32. But the bus speed does not get set, via the settings, resulting in a MCU-dependent default bus speed via the pre-scaler setting of 8, which is often much too high. I’m surprised my setup works at 9MHz, and I think if there were a BLDC driver and motor running chances are it wouldn’t work well any more.

the current code prevents the bus speed from being set correctly on STM32, so as a result, many people will have trouble with the current code under STM32 (as we see in this thread).

askuric, 2021-10-31:
Is this still an issue @runger1101001 ?

runger1101001, 2021-10-31:
No, this is fixed by removing the pre-scaler settings. I'm not sure where that came from originally. I suspect it was probably required in an older version of the STM32 platform, but is not required any more. Setting the SPI bus speed to the value given in the settings is definitely the better approach, IMHO.
So I think it can be closed.

closed 2021-10-31 by runger1101001