simplefoc/Arduino-FOC #84: Add code format checking

KaSroka, 2021-05-30:
This repo would benefit from keeping the code formatting more strict. Especially spaces around operators are all over the place.

What about adding https://github.com/marketplace/actions/clang-format-check github action?

runger1101001, 2021-05-31:
Hi @KaSroka 

Thanks for the suggestion, and the link. I'm of two minds:

1. I've never seen focusing on syntax actually really produce any tangible benefits. And personally, with limited time to spend on the project, I'd rather fix actual bugs or code cool new functionality. In terms of other users, I think they would benefit more if the time were spent on documentation rather than syntactic changes.

2. As a developer and computer scientist, there is a side to me that likes order, and I do try to keep the code I write "tidy" to my eyes. So in theory it would be nice if the code were all well formatted.

Ultimately I see this as Antun's call. Personally the 1. reasons would outweigh the 2. reason for me, and I wouldn't want to reformat all the code myself.

KaSroka, 2021-05-31:
@runger1101001 Reformatting is handled by clang-format when necessary.

Regarding 1. as a professional C programmer I can tell you that strict code formatting is a key to code maintainability. If it's ok I can create a PR with proposed changes so you can review it.