simplefoc/Arduino-FOC #89: Fail to compile for ESP32 error: '_ADC_VOLTAGE' was not declared in this scope

bgvsan, 2021-06-01:
from the latest changes:
[image]

bgvsan, 2021-06-01:
looks few esp32 test are excluded, maybe to address that issue we need to set a test for each platform supported.

askuric, 2021-06-02:
This is the dev branch?

bgvsan, 2021-06-02:
> This is the dev branch?

Yes

askuric, 2021-06-02:
Yeah, i think that the dev branch is properly broken at this moment. We'll change this soon though. 

You're right that this would be good to make. The only problem is the pinout and we would need to specify which examples to compile for which architecture. It's a bit of pain to make but i suppose it is possible and it makes sense for sure.

bgvsan, 2021-06-02:
> Yeah, i think that the dev branch is properly broken at this moment. We'll change this soon though. 
> 
> 
> 
> You're right that this would be good to make. The only problem is the pinout and we would need to specify which examples to compile for which architecture. It's a bit of pain to make but i suppose it is possible and it makes sense for sure.:D

I would have fixed if I know were we want to go but I we not been following any discussion recently. 
I can understand is a pain. I ve look at it and i still don't understood how. I need to learn a lot on this side.

No problem, will be a good to do for the future

askuric, 2021-06-10:
Ok, this has been fixed int the dev some time ago. And we've updated the github runners to compile for almost all  the target architectures so these things should not happen any more :)

closed 2021-06-10 by askuric