simplefoc/Arduino-FOC #90: questions about the support for stm32

candyriver, 2021-06-05:
I found that set PA8, PA9, PA10 as PWM pinout did not work.
I use the bluepill board and the code is just like this:
`
 BLDCDriver3PWM driver = BLDCDriver3PWM(PA8,PA9, PA10)
`
Other pins work well, such as  `BLDCDriver3PWM driver = BLDCDriver3PWM(PB13, PB14, PB15) ` or  `BLDCDriver3PWM driver = BLDCDriver3PWM(PB5, PB6, PB7) ` 

I am not sure the reason and could you help me? @owennewo

askuric, 2021-06-05:
Hey @candyriver,

Ive looked into it now and I cannot seem to be able to reproduce it. 
For me `PA8`, `PA9` and `PA10` work ;D

Could you tell us a bit more about your problem. 
What do you mean that the pins dont work, exactly?
Do you have any pwm output on any of the pins?
What is your Bluepill version?
What version of the stm32duino are you using?

candyriver, 2021-06-05:
Thanks for your response!
I will try another stm32 board and check it again. 
By the way, the version of stm32duino is 2.0.0.

askuric, 2021-06-05:
That might be the issue. Can you revert to 1.9.0?

I think that there is quite a bit of changes introduced in the 2.0 and I am not sure that we support it properly. 

We will need to look into this properly.

candyriver, 2021-06-06:
I tried another board and the problem disappeared!

closed 2021-06-06 by candyriver

langwadt, 2021-06-06:
fake stm32?