simplefoc/Arduino-FOC #96: Unable to get stepper turning

Theexternaldisk, 2021-06-29:
My issue is as follows. The motor rotates slightly during the initialisation but won't turn when using velocity or angle control. When a target is set via the terminal the amperage increases slowly untill the voltage_limit setpoint is reached. Then it will remain here and the stepper won't move.

My setup is using:
- Arduino mega
- VNH2SP30 motor shield from sparkfun
- optical rotary encoder (on the stepper shaft)

For the code I'm using the velocity_control example edited to work with the motor shield
```
StepperMotor motor = StepperMotor(50);//, 1.4);
StepperDriver2PWM driver = StepperDriver2PWM(       5,        7,        8,        6,        4,        9,           (int)A0,           (int)A1);
Encoder encoder = Encoder(18, 19, 1200);
```

Issues i've found:
- standard pwm frequency of 32khz is to high for the VNH2SP30 (max 20khz). Fixed this by changing the prescaler in the library to set the pwm frequency to 4khz
- setting the motor coil resistance will result in no output at all. Haven't looked into why

Any ideas?

Theexternaldisk, 2021-06-29:
I've created a separate issue for the pwm problem with the driver so others with the same issue can find it easier.

Theexternaldisk, 2021-07-05:
i've got the stepper to turn in openloop mode (velocity as well as angle).
pole pairs are configured correctly (verified by rotating 6.28 degrees).
can't reach speeds above 2 rad/s. any higher and it stops rotating.
closed loop stil does not work.
The initialisation fails on the PP check.

I've also been able to get it to work with the coil resistance set. Turns out the max current is defined at 0.2 A by default which resulted in a very low voltage limit. I set motor.current_limit to 2 instead and this works.

askuric, 2021-07-20:
Hey @Theexternaldisk ,

Were you able to get the closed loop running?
Did you test your sensor?

Did you try using the guide in: https://docs.simplefoc.com/example_from_scratch

askuric, 2021-08-25:
I'm going to close this one due to inactivity.

closed 2021-08-25 by askuric