simplefoc/Arduino-FOC #97: Standard PWM of 32 KHz to high for VNH2SP30

Theexternaldisk, 2021-06-29:
standard pwm frequency of 32khz is to high for the VNH2SP30 (max 20khz). 
Fixed this by changing the prescaler in the library to set the pwm frequency to 4khz

askuric, 2021-07-20:
We will attack this at the same time as #71 
The reason because it is fixed is die to the fact that you can only change the frequency with the prescaler which makes the granularity pretty high. So you can set either 32k, 16k, 8k,... 
And 16kHz is already audible, so we've decided to fix it to 32kHz for start.   
But it is true that we can let the user at least to decide if the prescaler if he wants. 
We will try to incorporate this in the next release.

askuric, 2022-12-22:
Included in the dev branch and will be in the next release. 
You can set either 32 or 4kHz pwm. 
Unfortunately the atmega chips do not support anything in the middle.

closed 2022-12-22 by askuric