simplefoc/Arduino-FOC #98: Unintended double promotions.

sDessens, 2021-07-01:
On platforms that have compiler support for doubles, such as the stm32, there are a lot of unintended float-to-double conversions in the code. I was able to reduce move() duration from 14us to 6us by eliminating just one of these unintended casts, and loopFOC() from 144us to 50us by replacing several. Almost all files have unintended promotions in them.

These instances can be easily found with -Wdouble-promotion and -Werror=double-promotion, there are a few hundred of them.

runger1101001, 2021-07-01:
There is a patch applied to the dev branch which fixes many of them, I believe.

We should keep this issue open and do another round on this topic, to make sure we have caught them all.

runger1101001, 2021-07-01:
Thank you for reporting this!

sDessens, 2021-07-01:
The dev branch fixes a few unintended casts, but there are still many left. Such as:

````
#define _round(x) ((x)>=0?(long)((x)+0.5):(long)((x)-0.5))
````
And:
````
#define _PWM_RANGE 4095.0// 2^12 -1 = 4095
_setPwm(pinA, _PWM_RANGE*dc_a, _PWM_RESOLUTION);
````
And:
````
duty_cycle1B = _constrain(abs(Ualpha)/voltage_power_supply,0.0,1.0);
````

I will report back later with a patch.

sDessens, 2021-07-01:
See pull request #100

sDessens, 2021-09-09:
Pull request merged

closed 2021-09-09 by sDessens