simplefoc/Arduino-FOC #99: __attribute__((weak)) doesn't seem to work as intended

sDessens, 2021-07-01:
Commit 0b7839f1e94356d4dde5757ddef412cce52d8331 broke pwm control on my machine and B-G431B. The offending change appears to be:

before: src/drivers/hardware_specific/generic_mcu.cpp 
```
#elif defined(_STM32_DEF_) // or stm32
/// do nothing
#else 
int _configure6PWM(...){ return -1; }
#endif
```

After: rc/drivers/hardware_specific/generic_mcu.cpp 
````
__attribute__((weak)) int _configure6PWM(...) {return -1;}
````

Although this implementation is declared weak, the linker still selects this implementation for the final executable, instead of the correct one in src/drivers/hardware_specific/stm32_mcu.cpp. This problem occurs with all weakly defined implementations in this file. Simply emptying the file returns everything to working order.

I'm observing very strange behavior when debugging this issue. Sometimes, removing just one of the functions from generic_mcu.cpp fixes the issue. Sometimes, it doesn't. This suggests we are depending on nondeterministic compiler behavior....

sDessens, 2021-07-01:
I'm observing a similar issue with all the weak symbols defined in src/current_sense/hardware_specific/generic_mcu.cpp

My platform is windows with toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)

runger1101001, 2021-09-06:
I'm actually having this problem with the current dev branch, STM32 and PlatformIO on MacOS...

runger1101001, 2021-09-06:
Note: I fixed this by adding

`lib_archive = false
`
to my platformio.ini as described in an unrelated issue here:
https://github.com/platformio/platform-ststm32/issues/205

sDessens, 2021-09-09:
I can verify runger's suggestion fixes this issue.

runger1101001, 2021-10-02:
Hmmm... SAMD21 has the same issue, and the same fix works there too.

Perhaps this __attribute__((weak)) business isn't ready for prime-time. I know it is far more elegant than the previous #ifdef based solution, and more scalable, but it has the severe disadvantage that 2 architectures (so far) require an additional compiler option in platformIO - lots of people will be caught out by that. At the very least we have to document this quite clearly...

closed 2022-02-18 by runger1101001