simplefoc/Arduino-FOC #102: Missing LPF when monitoring current in voltage mode

steph643, 2021-07-22:
All SimpleFOC data are monitored after being filtered. This allows users to tune the LPF settings. One exception though: in voltage mode, "current" values are sent to Serial without filtering. This is annoying when tuning with SimpleFOCStudio.

steph643, 2021-07-22:
Possible fix in FOCMotor.cpp line 107:
```C
    if(current_sense){
      if(torque_controller == TorqueControlType::foc_current) 
         c = current;
      else {
    	  c = current_sense->getFOCCurrents(electrical_angle);
          c.q = LPF_current_q(c.q);
          c.d = LPF_current_d(c.d);
      }
    }
```

askuric, 2021-07-23:
You're right, thanks for the suggestion!  
We need some cleaning up there, around what is shown in what mode in the monitor.
We will integrate, at least this fix, in the next release

closed 2021-09-16 by askuric