simplefoc/Arduino-FOC #105: defaut constructor missing

AurelienConil, 2021-07-29:
I created a class for the FOC part of my project : 
```
class BrushlessMotor{

    private :

    MagneticSensorSPI sensor;
    BLDCMotor motor;
    BLDCDriver3PWM driver;
}
```

Because these Class does not have defaut constructor, I get this error message : 
`src/main.cpp:36:19: error: no matching function for call to 'MagneticSensorSPI::MagneticSensorSPI()`

A solution of this problem is to use InitializerList ( https://www.geeksforgeeks.org/when-do-we-use-initializer-list-in-c/ )
According to this link : When do we use Initializer List in C++?
"3) For initialization of member objects which do not have default constructor: 
In the following example, an object “a” of class “A” is data member of class “B”, and “A” doesn’t have default constructor. 
Initializer List must be used to initialize “a”.

But at least, it should be probably easier to have defaut constructor for each class

last link : https://stackoverflow.com/questions/12328150/class-member-without-a-default-constructor

steph643, 2021-07-29:
@AurelienConil , it seems you're under the assumption that any class should have a default constructor. This assumption is wrong. See this explanation: https://www.quora.com/What-is-the-use-of-default-constructor.

AurelienConil, 2021-07-29:
Hi @steph643 , thanks for reply 
On the link you send to me: I found this : 
"If you will not provide any Constructor during the class definition. The the compiler will provide you an Zero Argument Constructor. That is known as Default Constructor. It is only used to initialize the class data members."
It seems that Arduino Compiler does not provide this Default Constructor. 
So what is the best pratice to implement FOC classes intro my own class ?

steph643, 2021-07-29:
A C++ compiler provides a default constructor when the developer doesn't provide any constructor.
MagneticSensorSPI *has* a constructor, so the compiler does not provide anything in this case.
To implement FOC classes intro your own class, use an initializer list as mentioned in your first post.

runger1101001, 2022-02-18:
Hey, I'm late to comment on this, but the correct way to do this is to provide a constructor for your class, which initializes the members. Something like this:

```
class BrushlessMotor{
    public:
     BrushlessMotor() : sensor(AS5048_SPI, 10), motor(7), driver(4,5,6) {};
    private:
    MagneticSensorSPI sensor;
    BLDCMotor motor;
    BLDCDriver3PWM driver;
};
```

Note that the values don't have to be hard-coded. They could be parameters of the constructor. So something like this:

```
class BrushlessMotor{
    public:
     BrushlessMotor(int pole_pairs, int pinA, int pinB, int pinC, int sensor_nCS) : sensor(AS5048_SPI, sensor_nCS), motor(pole_pairs), driver(pinA,pinB,pinC) {};
    private:
    MagneticSensorSPI sensor;
    BLDCMotor motor;
    BLDCDriver3PWM driver;
};
```

closed 2022-02-18 by runger1101001