simplefoc/Arduino-FOC #107: Can you give explanation of DRV8302 - High performance BLDC driver using Nucleo-64 F401RE example?

nonlinear1, 2021-08-17:
I am trying the DRV8302 - High performance BLDC driver using Nucleo-64 F401RE example, But I found no explanation about this example. Could someone knnow how to let Nucleo-64 F401RE run this example?
Think you very much!

askuric, 2021-08-17:
Hey @nonlinear1,

The code is exactly the same, and the connection diagram as well, as in https://docs.simplefoc.com/drv8302_example 
You are right that I should have precised it more in the docs though :smile:

nonlinear1, 2021-08-17:
You means the same? I think the F401RE the pin number or pin name is different from Arduino UNO! for pin configuration:#define   INH_A 9
#define   INH_B 10
#define   INH_C 11
#define   EN_GATE 8
#define   M_PWM 6 
#define   M_OC 5
#define   OC_ADJ 7
is the same? 
Here is the picture of stm32 version
[image]
[image]

askuric, 2021-08-17:
Yes, I guess you are right.
In the video I might have used different pins but it was mostly due to the convenience (to avoid the wire entaglement :smile: ) .
You can use exactly the same pins as for the Arduino UNO with your Nucleo-64 Board.

The only issue that I had sometimes was using the pwm on the pin `11`, so I'd suggest you to use the pin `13` instead pinout like this:
```
#define   INH_A 9
#define   INH_B 10
#define   INH_C 13
#define   EN_GATE 8
#define   M_PWM 6 
#define   M_OC 5
#define   OC_ADJ 7
```

nonlinear1, 2021-08-17:
Thank you Askuric! You are a kind-hearted and give me great help :)

nonlinear1, 2021-08-17:
I am a newer for Arduino. I want to know your pin name definition.
for the specification of  Nucleo-64 F401RE is as follow: 
[image]
your definition of pin, for exampe: pin 11 means D11 as in the specification of Nucleo-64 F401RE?

askuric, 2021-08-17:
Yes that is exactly it, sometimes the piout diagrams get useful as well. You can find the by googling relativey easity: 
Here is one of the nucleo-64 F401RE

[image]

nonlinear1, 2021-08-17:
Thank you very much!

nonlinear1, 2021-08-19:
@askuric Can I run the example of DRV8302 without the Encoder?

askuric, 2021-10-31:
Hey @nonlinear1,
I'm going to close this issue. If you have some more questions feel free to reopen it or create a new one. 
There is also the community.simplefoc.com which is very active and can give you some more responsive solutions 😄

closed 2021-10-31 by askuric