simplefoc/Arduino-FOC #109: will SimpleFOC support ESP32S2

juedongli, 2021-08-22:
Arduino: 1.8.15 (Windows 7), Board: "ESP32S2 Dev Module, UART0, UART0, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), QIO, 80MHz, 4MB (32Mb), 921600, None"

d:\Documents\Arduino\libraries\Simple_FOC\src\drivers\hardware_specific\esp32_mcu.cpp:6:10: fatal error: soc/mcpwm_reg.h: No such file or directory

 #include "soc/mcpwm_reg.h"

          ^~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1

Error compiling for board ESP32S2 Dev Module.

askuric, 2021-08-24:
Hey @juedongli ,
Thanks for reporting this. What I'd the esp32 package version you're using?
This seems to be a bit of a strange issue, from the esp32 specific code that i was able to find online there shouldn't be any real difference in between esp32 and esp32s2. Al least for the owl generation.

jkirsons, 2021-09-04:
The ESP32 S2 does not have MCPWM listed in it's block diagram in the datasheet.  So I don't think it has it.

The ESP32 S3 lists MCPWM, so I think we'll have to wait for the S3.

juedongli, 2021-09-11:
I'm using ESP32-DevKitC-1 (ESP32-WROVER-E)

askuric, 2022-02-03:
esp32 s2 and c3 which do not have the mcpwm are supported starting from v2.2.1

closed 2022-02-03 by askuric