simplefoc/Arduino-FOC #112: ESP32 position motion control example with magnetic sensor: SPI pins

juedongli, 2021-09-05:
ESP32 position motion control example with magnetic sensor
"""
// SPI Magnetic sensor instance (AS5047U example)
// MISO 12
// MOSI 9
// SCK 14
// magnetic sensor instance - SPI
MagneticSensorSPI sensor = MagneticSensorSPI(AS5147_SPI, 10);
"""

ESP32-DevKitC-1(ESP32-WROVER-E) has no pin10
HSPI_SS  pin is 15

  #define HSPI_MISO   12
  #define HSPI_MOSI   13
  #define HSPI_SCLK   14
  #define HSPI_SS     15

the example  does not work

jkirsons, 2021-09-10:
Hi, I don't think it's possible to provide 1 example that directly works for every Dev Kit.  Even different ESP32 package variants expose different pins.

You can remap the pins used in the sensor constructor and SPIClass begin method:
```
MagneticSensorSPI sensor = MagneticSensorSPI(<MagneticSensorSPIConfig_s>, <CS pin>);
SPI.begin(<CLK pin>, <MISO pin>, <MOSI pin>, <CS pin>);
sensor.init(&SPI);
...
```

juedongli, 2021-09-11:
it works. thank you @jkirsons 

`
#include <SimpleFOC.h>

// MagneticSensorSPI(int cs, float _cpr, int _angle_register)
// config           - SPI config
//  cs              - SPI chip select pin 
//MagneticSensorSPI sensor = MagneticSensorSPI(AS5147_SPI, PA15);
MagneticSensorSPI sensor = MagneticSensorSPI(15, 14, 0x3FFF);

// HSPI 
// CS 15
// SCK 14
// MOSI 13
// MISO 12
// these are valid pins (mosi, miso, sclk) for 2nd SPI bus on storm32 board (stm32f107rc)
//SPIClass SPI_2(13, 12, 14);
// SPI_2(PB13, PB12, PB14);

#define HSPI_MISO   12
#define HSPI_MOSI   13
#define HSPI_SCLK   14
#define HSPI_SS     15

SPIClass * hspi = NULL;

void setup() {
  // monitoring port
  Serial.begin(115200);

  hspi = new SPIClass(HSPI);
  hspi->begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS); //SCLK, MISO, MOSI, SS
  //pinMode(HSPI_SS, OUTPUT); //HSPI SS

  // initialise magnetic sensor hardware
  //sensor.init(&SPI_2);
  sensor.init(hspi);

  Serial.println("Sensor ready");
  _delay(2000);
}

void loop() {
  // display the angle and the angular velocity to the terminal
  Serial.print(sensor.getAngle());
  Serial.print("\t");
  Serial.println(sensor.getVelocity());
}

`

askuric, 2021-09-16:
Hey guys,
Sorry for the delay, I'm happy you've made it work!

I've updated the esp32 example to use 15 as CS pin and I've added the new example to the spi sensor only for the esp32 which shows how to setup the HSPI spi bus.
Its currently in the dev branch but it will go out with the new release.

closed 2021-09-16 by askuric