simplefoc/Arduino-FOC #128: STM32G4xx current sensing can't compile (PlatformIO)

runger1101001, 2021-11-01:
Not sure if this issue is platformIO-specific, or also occurs in ArduinoIDE...

It is found by forum user dikafoc, and described here: https://community.simplefoc.com/t/b-g431b-esc1-beginner-guide-i2c-guide/515/46?u=runger
I have confirmed it also occurs on my MacBook.

Problem is that there are missing includes for hal api, so the structures used in the header file stm32g4_hal.h aren't declared.

runger1101001, 2021-11-01:
Fix merged, but should be tested in ArduinoIDE as well as PlatformIO before I close this.

st157322, 2021-11-30:
I have tested it on Arduino IDE and PIO and it does not work.

on Arduino i get:

`/Users/lukas/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: libraries/Arduino-FOC/current_sense/hardware_specific/stm32g4_mcu.cpp.o: in function `_configureOPAMP(OPAMP_HandleTypeDef*, OPAMP_TypeDef*)':
stm32g4_mcu.cpp:(.text._Z15_configureOPAMPP19OPAMP_HandleTypeDefP13OPAMP_TypeDef+0x22): undefined reference to `HAL_OPAMP_Init'
/Users/lukas/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: libraries/Arduino-FOC/current_sense/hardware_specific/stm32g4_mcu.cpp.o: in function `_configureADCInline(int, int, int)':
stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x66): undefined reference to `HAL_OPAMP_Start'
/Users/lukas/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x6c): undefined reference to `HAL_OPAMP_Start'
/Users/lukas/Library/Arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.2.1-1.1/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x72): undefined reference to `HAL_OPAMP_Start'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Discovery.`

and on PIO i get a similar error:

`/Users/lukas/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/disco_b_g431b_esc1/lib84b/Simple FOC@src-b4927adb2d82a99aa14a5e549dad1b2c/current_sense/hardware_specific/stm32g4_mcu.cpp.o: in function `_configureOPAMP(OPAMP_HandleTypeDef*, OPAMP_TypeDef*)':
stm32g4_mcu.cpp:(.text._Z15_configureOPAMPP19OPAMP_HandleTypeDefP13OPAMP_TypeDef+0x22): undefined reference to `HAL_OPAMP_Init'
/Users/lukas/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/disco_b_g431b_esc1/lib84b/Simple FOC@src-b4927adb2d82a99aa14a5e549dad1b2c/current_sense/hardware_specific/stm32g4_mcu.cpp.o: in function `_configureADCInline(int, int, int)':
stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x66): undefined reference to `HAL_OPAMP_Start'
/Users/lukas/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x6c): undefined reference to `HAL_OPAMP_Start'
/Users/lukas/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: stm32g4_mcu.cpp:(.text._Z19_configureADCInlineiii+0x72): undefined reference to `HAL_OPAMP_Start'
collect2: error: ld returned 1 exit status
*** [.pio/build/disco_b_g431b_esc1/firmware.elf] Error 1`

st157322, 2021-11-30:
Yes, i used the dev branch and did a clean build.

I managed to get it work with PIO by using
`    -D HAL_ADC_MODULE_ONLY
    -D HAL_OPAMP_MODULE_ENABLED`
as additional Buildflag

askuric, 2022-05-02:
Is this fixed in the master then or do we still have an issue with the build flags?

lurenaud, 2022-05-17:
There still have the issue without the additional build flag.
Also, only the `-D HAL_ADC_MODULE_ONLY` is necessary. (otherwise there are a lot of redefinition warning for the `HAL_OPAMP_MODULE_ENABLED` flag)

runger1101001, 2023-03-21:
I think this issue is fixed with the refactoring of the current sensing code?

runger1101001, 2023-11-28:
As many users are now using the STM32G4 current sensing, it has been fixed and I will now close this issue.

closed 2023-11-28 by runger1101001