simplefoc/Arduino-FOC #133: Is it possible to use the Arduino SimpleFOC Library with an ESC?

thankyoucubed, 2021-11-09:
Hi y'all, I apologize in advance if this has been asked before/is a silly question - I'm quite new to this area of motor control.

I was curious if it's possible to use this ESC (https://www.rcelectricparts.com/100a-rc-brushless-motor-electric-speed-controller-esc-4a-ubec-with-xt60--35mm-bullet-plugs.html) with this BLDC motor (https://www.scorpionsystem.com/catalog/boat/motor_3/is-34_series/is_3435_1800/) with the FOC code? I'm hoping to use this setup to do torque control mode on the motor down the road.

I'm having a very difficult time finding a motor driver for this particular motor. I see that the pinout that works with the code requires the 3 PWM pins (1 pin for each phase of the motor), but an ESC only utilizes the 1 PWM pin as the signal to translate to the motor. Is it possible to use an ESC still?

Any help would be greatly appreciated, thanks

runger1101001, 2021-11-10:
Hey @thankyoucubed ,

Welcome to SimpleFOC - please consider using our forum for this kind of question: [community.simplefoc.com](https://community.simplefoc.com/) - there's quite an active community, and there could be others who have ideas about this.

To answer your question: at the moment this is not possible. SimpleFOC would have be running on the ESC to enable FOC control of the motor, or the ESC would have to be "simpler" and offer direct control of its FET switches to the outside. Neither is the case here. Typically these ESCs run some kind of simple 6-step commutation control software. They have input for RC-receiver, and process this into the motor control by themselves. Their MCUs aren't powerful enough to run FOC, and they don't have the inputs for the additional position sensor needed.

The motor looks like a fairly standard BLDC in-runner. With some tuning of the PID-parameters, a driver that can handle higher currents, and a fast position sensor you should be able to get it working well with FOC control.

You can't do the FOC with a 1-PWM ESC... you'll need a BLDC motor driver board that does 3-PWM or 6-PWM. Check the SimpleFOC documentation and the forum for some ideas what can work...

runger1101001, 2021-11-15:
I'm sorry to hear the forum signup didn't work as expected... would you mind sharing your username and email you signed up with on the forum, then I can check what's going on there?

runger1101001, 2022-10-30:
I'm going to close this for now. Do let us know if you have other questions!

closed 2022-10-30 by runger1101001