simplefoc/Arduino-FOC #138: PWM driver doesn't work with Platformio

albertomercurio, 2021-11-22:
Hello to everyone,

I'm working with the STM32 BluePill board. I'm actually able to reproduce every voltages on each line with the following example (with Arduino IDE)

```
#include <Arduino.h>

#define PWM_A_H PA8
#define PWM_A_L PA7
#define PWM_B_H PA9
#define PWM_B_L PB0
#define PWM_C_H PA10
#define PWM_C_L PB1
#define DRV_EN_PIN PB10

BLDCDriver6PWM driver = BLDCDriver6PWM(PWM_A_H, PWM_A_L, PWM_B_H, PWM_B_L, PWM_C_H, PWM_C_L, DRV_EN_PIN);

void setup() {
  // pwm frequency to be used [Hz]
  // for atmega328 fixed to 32kHz
  // esp32/stm32/teensy configurable
  driver.pwm_frequency = 50000;
  // power supply voltage [V]
  driver.voltage_power_supply = 12;
  // Max DC voltage allowed - default voltage_power_supply
  driver.voltage_limit = 12;
  // daad_zone [0,1] - default 0.02f - 2%
  driver.dead_zone = 0.05f;

  // driver init
  driver.init();

  // enable driver
  driver.enable();
}

void loop() {
  driver.setPwm(3,6,5);
}
```

But when I try to work on Platformio in VS Code there is no voltage output (of course I added `#include <SimpleFOC.h>`). The code can be succesfull compliled and uploaded, but nothing happens. The code is able to blink the builtin led, is able to use the USART2 interface, and other things.

askuric, 2021-11-23:
This seems very strange. I am not using the platformio so I am maybe not the best person to respond. The issues that I see that could happen is that platforio somehow does not use stm32duino, is that possible?
What is the version of the stm32duino you're using?
The other issue could be the issue resolved recently in:
https://github.com/simplefoc/Arduino-FOC/issues/99#issuecomment-913887085

albertomercurio, 2021-11-23:
Perfect, I solved with `lib_archive = false` option! Thank you.

closed 2021-11-23 by albertomercurio

askuric, 2021-11-23:
Cool, I'm happy it helped.
By the way we have a brand new Platformio installation guide in our docs made by @runger1101001 
https://docs.simplefoc.com/library_platformio

BobBaoBao, 2022-06-17:
I also incounter this problem. I add lib_archive = false in my platformio.ini, but it still doesn't work.
here is my code

#include <Arduino.h>
#include <SimpleFOC.h>

BLDCDriver6PWM driver = BLDCDriver6PWM(PA8, PB13, PA9, PB14, PA10, PB15, PB12);

void setup() {

  // pwm frequency to be used [Hz]
  // for atmega328 fixed to 32kHz
  // esp32/stm32/teensy configurable
  driver.pwm_frequency = 50000;
  // power supply voltage [V]
  driver.voltage_power_supply = 24;
  // Max DC voltage allowed - default voltage_power_supply
  driver.voltage_limit = 12;
  // daad_zone [0,1] - default 0.02f - 2%
  driver.dead_zone = 0.05f;

  // driver init
  driver.init();

  // enable driver
  driver.enable();
}

void loop() {
    // setting pwm
    // phase A: 3V
    // phase B: 6V
    // phase C: 5V
    driver.setPwm(3,6,5);
}

here is the platformio.ini
[env:bluepill_f103c8_128k]
platform = ststm32
board = bluepill_f103c8_128k
; board = genericSTM32F103CB
; board_build.core = maple
framework = arduino
debug_tool = stlink
upload_protocol = stlink
lib_deps = askuric/Simple FOC@^2.2.2
lib_archive = false
monitor_speed = 115200
build_flags = 
	-D PIO_FRAMEWORK_ARDUINO_SERIAL_DISABLED
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC

I solved the PWM wave problem according to this web:https://community.platformio.org/t/issue-with-stm32-blue-pill-and-pwmwrite-function/18745

But the simpleFOC BLDC_Driver_6PWM_alone still can't work