simplefoc/Arduino-FOC #147: Compile Error with ESP32...

TSJim, 2022-01-03:
I'm trying to run the following sketch on an ESP32 (WEMOS LOLIN32):

```Simple_FOC/examples/utils/driver_standalone_test/bldc_driver_6pwm_standalone/bldc_driver_6pwm_standalone.ino```

However, I am getting compiler errors as shown below when I try to compile this sketch or any of the other examples in the library.  The last lines are:

```
exit status 1
Error compiling for board WEMOS LOLIN32.
```

and above that are all the following error messages:

***

```
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp: In function 'void _configureTimerFrequency(long int, mcpwm_dev_t*, mcpwm_unit_t, float)':
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:127:22: error: 'volatile union mcpwm_clk_cfg_reg_t' has no member named 'prescale'
   mcpwm_num->clk_cfg.prescale = 0;
                      ^~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:143:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[0].period.prescale = prescaler;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:144:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[1].period.prescale = prescaler;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:145:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[2].period.prescale = prescaler;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:148:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[0].period.period = resolution_corrected;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:149:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[1].period.period = resolution_corrected;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:150:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[2].period.period = resolution_corrected;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:152:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[0].period.upmethod = 0;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:153:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[1].period.upmethod = 0;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:154:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'period'
   mcpwm_num->timer[2].period.upmethod = 0;
                       ^~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:165:73: warning: 'esp_err_t mcpwm_sync_enable(mcpwm_unit_t, mcpwm_timer_t, mcpwm_sync_signal_t, uint32_t)' is deprecated: please use mcpwm_sync_configure() instead [-Wdeprecated-declarations]
   mcpwm_sync_enable(mcpwm_unit, MCPWM_TIMER_0, (mcpwm_sync_signal_t)1, 0);
                                                                         ^
In file included from /Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:5:
/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:838:11: note: declared here
 esp_err_t mcpwm_sync_enable(mcpwm_unit_t mcpwm_num, mcpwm_timer_t timer_num, mcpwm_sync_signal_t sync_sig,
           ^~~~~~~~~~~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:165:73: warning: 'esp_err_t mcpwm_sync_enable(mcpwm_unit_t, mcpwm_timer_t, mcpwm_sync_signal_t, uint32_t)' is deprecated: please use mcpwm_sync_configure() instead [-Wdeprecated-declarations]
   mcpwm_sync_enable(mcpwm_unit, MCPWM_TIMER_0, (mcpwm_sync_signal_t)1, 0);
                                                                         ^
In file included from /Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:5:
/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:838:11: note: declared here
 esp_err_t mcpwm_sync_enable(mcpwm_unit_t mcpwm_num, mcpwm_timer_t timer_num, mcpwm_sync_signal_t sync_sig,
           ^~~~~~~~~~~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:166:73: warning: 'esp_err_t mcpwm_sync_enable(mcpwm_unit_t, mcpwm_timer_t, mcpwm_sync_signal_t, uint32_t)' is deprecated: please use mcpwm_sync_configure() instead [-Wdeprecated-declarations]
   mcpwm_sync_enable(mcpwm_unit, MCPWM_TIMER_1, (mcpwm_sync_signal_t)1, 0);
                                                                         ^
In file included from /Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:5:
/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:838:11: note: declared here
 esp_err_t mcpwm_sync_enable(mcpwm_unit_t mcpwm_num, mcpwm_timer_t timer_num, mcpwm_sync_signal_t sync_sig,
           ^~~~~~~~~~~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:166:73: warning: 'esp_err_t mcpwm_sync_enable(mcpwm_unit_t, mcpwm_timer_t, mcpwm_sync_signal_t, uint32_t)' is deprecated: please use mcpwm_sync_configure() instead [-Wdeprecated-declarations]
   mcpwm_sync_enable(mcpwm_unit, MCPWM_TIMER_1, (mcpwm_sync_signal_t)1, 0);
[... 11 lines cut ...]
           ^~~~~~~~~~~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:167:73: warning: 'esp_err_t mcpwm_sync_enable(mcpwm_unit_t, mcpwm_timer_t, mcpwm_sync_signal_t, uint32_t)' is deprecated: please use mcpwm_sync_configure() instead [-Wdeprecated-declarations]
   mcpwm_sync_enable(mcpwm_unit, MCPWM_TIMER_2, (mcpwm_sync_signal_t)1, 0);
                                                                         ^
In file included from /Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:5:
/Users/jim/Library/Arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:838:11: note: declared here
 esp_err_t mcpwm_sync_enable(mcpwm_unit_t mcpwm_num, mcpwm_timer_t timer_num, mcpwm_sync_signal_t sync_sig,
           ^~~~~~~~~~~~~~~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:169:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'sync'
   mcpwm_num->timer[0].sync.out_sel = 1;
                       ^~~~
/Users/jim/Sync/Documents_Shared/Arduino_Shared/libraries/Simple_FOC/src/drivers/hardware_specific/esp32_mcu.cpp:171:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'sync'
   mcpwm_num->timer[0].sync.out_sel = 0;
                       ^~~~
```

***

I can compile it just fine on an Arduino Nano, and I can compile other sketches on the ESP32 (the WEMOS LOLIN32 is the board I use nearly all the time and buy them in 100 piece lots once or twice a year).

I have seen other threads here that are similar concerning the support of the ESP32 and compiler errors, but no solutions.

Is the ESP32 supported by the SimpleFOC library?

Thank you!

TSJim, 2022-01-03:
Hey @askuric 

I appreciate your fast reply!

On your advice, I downgraded to version 2.0.0, restarted the Arduino IDE, and opened up the bldc_driver_6pwm_standalone.ino sketch.  However, I am still getting the same compiler errors as I posted above.

What can I try next?

Thank you!

samguns, 2022-01-09:
I think this 2.0.2 adaptation issue has been addressed by PR[#149](https://github.com/simplefoc/Arduino-FOC/pull/149).

askuric, 2022-01-10:
Hey @samguns,
yep the dev branch has this issue resolved. We will make the v2.2.1 release soon so that it enters the Arduino library manager as well. 
Thanks for the PR!

askuric, 2022-02-03:
This has beed resolved with the release v2.2.1 and the pr #149

closed 2022-02-03 by askuric