simplefoc/Arduino-FOC #148: STM32G4 compile problem

runger1101001, 2022-01-04:
There is a problem in stm32g4_mcu.cpp in the current sense:

in the function:

```
float _readADCVoltageInline(const int pin){
  uint32_t raw_adc = 0;
  if(pin == PA2)  // = ADC1_IN3 = phase U (OP1_OUT) on B-G431B-ESC1
    raw_adc = adcBuffer1[1];
  else if(pin == PA6) // = ADC2_IN3 = phase V (OP2_OUT) on B-G431B-ESC1
    raw_adc = adcBuffer2[0];
  else if(pin == PB1) // = ADC1_IN12 = phase W (OP3_OUT) on B-G431B-ESC1
    raw_adc = adcBuffer1[0];
  return raw_adc * _ADC_CONV;
}
```

The core of the problem is that the whole file is specific to the B-G431-ESC1 board, but is "masquerading" as a driver for the STM32G4. The #ifdefs protecting the compilation should specifically target the B-G431-ESC1 board, and not the G4 series.

Quite apart from the fact that this implementation with its hard-coded pins won't work for people using other G4 boards anyway, it also forces dependence on the OPAMP module and in this particular instance, prevents compilation on some boards due to its hard-coded reference to pin PB1.... PB1 is not used on the Nucleo G431 board for example, and therefore not defined in the STM core's headers as a pin. :-(

A short order solution should be to make the #ifdefs specifically target the ESC1 board. In the medium term, we should refactor the current sensing as discussed (work is ongoing :-) )

askuric, 2022-05-02:
This should be resolved with the new release?

askuric, 2022-10-15:
This is deprecated from v2.2.2

closed 2022-10-15 by askuric