simplefoc/Arduino-FOC #159: Investigate AS5048A and MagneticSensorSPI

runger1101001, 2022-02-13:
Two recent problems in the forum related to AS5048A and MagneticSensorSPI:

> This turned out to be the key: when I changed my code to use the 5147 default config it immediately worked. I’m super puzzled over this though, because as far as I understand from checking out MagneticSensorSPI.cpp, the as5048 default config is literally a copy of the as5147 default config. They do clearly behave differently though, so there must be something going on.

from: https://community.simplefoc.com/t/challenges-going-faster-than-certain-rpm/1688/5

and

> I’ve just done some more tests and haven’t witnessed the strange behavior from last month.
> All I see now is that my AS5048A:
> works correctly with MagneticSensorAS5048A(10)
> doesn’t work at all with MagneticSensorSPI(10, 14, 0x3FFF) (hangs in SPI communication)

from: https://community.simplefoc.com/t/issue-with-as5048a-magnetic-sensor/983/10

runger1101001, 2022-06-19:
Reported today also on Discord:
[image]

This was on an STM32F765IIK6 MCU, again with AS5048A.

PumaFPV, 2022-06-19:
Here is what it looks like with MagneticSensorAS5048A
[image]
And with MagneticSensorSPI
[image]
First of all SPIsettings are different so this doesnt help

runger1101001, 2022-08-29:
[image]

runger1101001, 2022-08-29:
Another report on Discord, where switching from AS5048 settings to AS5047 settings worked... I think it might have to do with the extern declaration and the order of initialisation... it would be an explanation.