simplefoc/Arduino-FOC #162: soc/soc_caps.h: No such file or directory

zhangck13, 2022-03-19:
I'm using platformIO and get the error below:

`.pio/libdeps/lolin_d32/Simple FOC/src/drivers/hardware_specific/esp32_ledc_mcu.cpp:21:26: fatal error: soc/soc_caps.h: No such file or directory`

I'm aware of this answer: [](https://community.simplefoc.com/t/compilation-error/1700). But, how could I make things work again after moving to the new SimpleFOC ver 2.2.1?

 Thank you!

askuric, 2022-03-20:
Hey @zhangck13,

We have transitioned to the esp32 package version 2.0.1+ and in order to use the v2.2.1 you need to install the earliest version of the esp32 arduino package.
Which version of the ESP32 support package do you have?

The earlier versions will not work. You might be able to remove the `esp32_ledc_mcu.cpp` file and compile the code but you will not be able to make it work because the needed defines will not be set for MCPWM. And even if they would be set, the mcpwm API has changed and it would not compile. 

Here is how to update your esp32 package version with Arduino IDE and platformio : https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#

zhangck13, 2022-03-20:
@askuric Thank you very much for your help! I successfully updated my esp32 package according to the instructions that you posted and that solved the problem.

closed 2022-03-20 by zhangck13