simplefoc/Arduino-FOC #168: Any experience with nanoFOC?

rocketrobin777, 2022-04-13:
Dear simpleFOC community,

I thought about "how to start" with simpleFOC and ordered the Community-Board directly from simplefoc.com.
I also found a board, called nanoFOC on [Aliexpress](https://de.aliexpress.com/item/1005003337066069.html). 
Does anyone have experience with this board?
For my application it would be a nice hardware to implement the function I want to test. But the [documentation](http://lgn.life/simplefoc%e5%bc%80%e5%8f%91%e5%a5%97%e8%a3%85%e5%9f%ba%e7%a1%80%e6%95%99%e7%a8%8b/) from the seller is in chinese and google translater makes the job sufficient to get first information, but:
I'm not sure if I'm able to flash my own code on it. As I see in the description, there are different options to choose a microcontroller. And I also guess that the AS5600 is onboard. 
Does Anyone have better docs about it?
Thanks in advance.

Robin

runger1101001, 2022-04-13:
Dear @rocketrobin777 

I haven't actually seen this design before, it looks like a very nice small design! I'll add it to the list :-)

Without schematics or further documentation, I'm afraid I can't give you any more information than you see on AliExpress...

Two things to try:
- Make a post in the SimpleFOC forums - chances are the designer of this board is on the forum, or perhaps one of our Chinese users can help you to get more information
- Contact the seller on AliExpress... often for this kind of thing the seller is also the maker and can give you the details you need...

If you decide to buy one, and try it out, it would be great to hear about your experiences. Personally the price is a bit high for me to order it just to try it out.

If it is ok with you I will close this ticket - we use GitHub more for bugs and code improvements, and this kind of question is better posed in our forums...

closed 2022-04-13 by runger1101001

rocketrobin777, 2022-04-14:
Thanks for your recommendation. Sure, it's OK to close the issue. I will make a new thread on simpleFOC.