simplefoc/Arduino-FOC #171: BLDCMotor.cpp typo?

bobbasham, 2022-04-19:
Is the check against open loop missing curly braces?

void BLDCMotor::move(float new_target) {

  // downsampling (optional)
  if(motion_cnt++ < motion_downsample) return;
  motion_cnt = 0;

  // shaft angle/velocity need the update() to be called first
  // get shaft angle
  // TODO sensor precision: the shaft_angle actually stores the complete position, including full rotations, as a float
  //                        For this reason it is NOT precise when the angles become large.
  //                        Additionally, the way LPF works on angle is a precision issue, and the angle-LPF is a problem
  //                        when switching to a 2-component representation.
  **if( controller!=MotionControlType::angle_openloop && controller!=MotionControlType::velocity_openloop ) 
    shaft_angle = shaftAngle(); // read value even if motor is disabled to keep the monitoring updated but not in openloop mode
  // get angular velocity 
  shaft_velocity = shaftVelocity(); // read value even if motor is disabled to keep the monitoring updated**

  // if disabled do nothing
  if(!enabled) return;

askuric, 2022-04-20:
No there is no need for curly braces there.
It is important not to use the sensor angle when in openloop mode. The velocity value is not used in the openloop algorithm (or is calculated each time in the openloop functions) the velocity line does not make any difference.

I mean we could enclose both in curly braces as well, if that's what you're asking. 
But the algorithm will not be affected.

askuric, 2022-05-02:
Hey @bobbasham,
If you have any more questions dont hesitate to make another issue, you can also post it to the the community or discord server.
I am closing this one :D

closed 2022-05-02 by askuric