simplefoc/Arduino-FOC #172: initFOC with MT6701Sensor get wrong pole pair value and electrical zero offset

caishanli, 2022-04-20:
I use mt6701 sensor to replace with as5600, and get wrong pole pair value and electrical zero offset when call initFOC with default params, and the motor power(about 5W) is very high. I have to manual test motor's electrical zero offset to reduce power(1.5W).

mt6701 code is [mt6701_sensor.h](https://github.com/scottbez1/smartknob/blob/master/firmware/src/mt6701_sensor.h) and [mt6701_sensor.cpp](https://github.com/scottbez1/smartknob/blob/master/firmware/src/mt6701_sensor.cpp), and this project also have the same [problem](https://github.com/scottbez1/smartknob/blob/master/firmware/src/motor_task.cpp#L45).

runger1101001, 2022-04-20:
Hi Scott, love the SmartKnob project!

I must admit I don't exactly see what's going on here... in particular, I can't quite make sense of your code based on the datasheet...

The MT6701 looks like a nice sensor. I think I will order some and write a specific driver for it.
But in the meantime, from the looks of things it should not be hard to support it.

Your code looks like it is applying correction to the raw sensor value by doing something like an LPF on two orthogonal component vectors of the output? I don't fully get it, but I assume it makes sense?

First things to check from my point of view:

- does the raw value returned by the sensor make sense if you print it?
- make a simple sensor test program without any of the other stuff, which just initialises the sensor, and calls sensor.update() and sensor.getAngle() --> if you print those angle values, are they correct?
- comparing with the corrected value, does it make sense? Is the corrected value close to the original?
- is the performance better if you use the raw value or the corrected value?
- do the values vary correctly between 0 and 2PI as you turn through a full rotation?

- what MCU are you using?

caishanli, 2022-04-21:
> Hey, I love the SmartKnob project!
> 
> I must admit I don't exactly see what's going on here... in particular, I can't quite make sense of your code based on the datasheet...
> 
> The MT6701 looks like a nice sensor. I think I will order some and write a specific driver for it. But in the meantime, from the looks of things it should not be hard to support it.
> 
> Your code looks like it is applying correction to the raw sensor value by doing something like an LPF on two orthogonal component vectors of the output? I don't fully get it, but I assume it makes sense?
> 
> First things to check from my point of view:
> 
> * does the raw value returned by the sensor make sense if you print it?
> * make a simple sensor test program without any of the other stuff, which just initialises the sensor, and calls sensor.update() and sensor.getAngle() --> if you print those angle values, are they correct?
> * comparing with the corrected value, does it make sense? Is the corrected value close to the original?
> * is the performance better if you use the raw value or the corrected value?
> * do the values vary correctly between 0 and 2PI as you turn through a full rotation?
> * what MCU are you using?

Yes in my test, mt6701 is better than as5600, and cheaper.
I use simpleFocStudio to show the angle curve of mt6701 sensor, and make a full rotation, the value changed about 2pi.
my mcu is esp32.
the estimated pole pair is 18.76(every not same, but very close), and the real is 11
and the estimated electrical zero offset is 5.85, my tested value is 3.7

runger1101001, 2022-04-23:
I have ordered some MT6701 sensors and I will try it out. I can't really say what might be wrong based on the information given...
In my experience, the magnetic sensors can be fussy to set up in terms of getting the communication working, and many people have trouble with the magnets. It absolutely has to be a diametrically magnetised magnet, and there are different magnet qualities. The placement of the sensor is critical, it has to be the right distance and very well centred to get good results.
Often a sensor will "kind of work" with a bad magnet or misaligned position, it seems to work when turned by hand, but then fails to work when turned by the motor (which generally has much faster movement and acceleration).
Another common problem is bad cables - SPI is not intended for cables, and if the cables are too long, or there is a lot of electromagnetic noise (like from a motor, or phase cables) then the communication can be unreliable.

I see that you read the sensor's status bits in your code... what does the sensor think about the magnetic field strength, did you check the status bits?

But once you get past these problems and are able to read the values, the sensors normally work pretty reliably.